fluidd-core/fluidd #158: I get an INVALID error

ChrisYOLO, 2021-02-22:
For some reason i get this error displayed below. I haven't changed anything in my setup and this has happend to me twice. Printer is connected.
[image]
[moonraker (1).log](https://github.com/cadriel/fluidd/files/6024491/moonraker.1.log)
[klippy (2).log](https://github.com/cadriel/fluidd/files/6024492/klippy.2.log)

ChrisYOLO, 2021-02-22:
I saw that you closed the previous one but i didn't get to reply. No the printer waws not updating and i have tried manually updating everyhitng, still not working :/

cadriel, 2021-02-22:
Please, stop creating issues.

You can still reply to a closed issue.

This is not directly a Fluidd issue. The state of those updates comes from Moonraker. It's telling you that the repositories are invalid. Please ensure they're not forks - and that you haven't touched any of the files inside those folders.

Also - feel free to reach out in Discord, where this would be more easily diagnosed.

closed 2021-02-22 by cadriel