fluidd-core/fluidd #159: Jobs/Config on Dashboard - Buttons not aligned

jrdnlc, 2021-02-24:
[image]
[image]

cadriel, 2021-03-03:
I can't replicate this. Can you provide further information on your browser, and operating system? Also - what screen resolution / size is your browser set at?

jrdnlc, 2021-03-06:
OSX and Safari. It happens on latest version of iOS as well[image].

cadriel, 2021-03-06:
Hrm, I'm on OSX, and have tested in Safari - and don't have this issue.

Do you have your browsers set with a zoom? Can you try hitting CMD-0, or go into the menu and ensure zoom is set to 0 / off?

jrdnlc, 2021-03-06:
Zoom is at 0. I can replicate on iOS so I highly doubt that’s it

cadriel, 2021-03-06:
Ah I can see it on IOS. That's enough for me to better diagnose, cheers.

cadriel, 2021-03-15:
This will be fixed in the next release.

closed 2021-03-15 by cadriel