fluidd-core/fluidd #163: WebRTC does not show up but element does

icebal, 2021-03-02:
using a webRTC streamer server, setting the camera as IP Camera, and the url as:
http://octopi:8000/webrtcstreamer.html?video=printer
the video does not show up, but if i right click the Camera box on the dashboard and copy video location, it works fine separately.
the video is the only <video> tag on that url as well. 
the WebRTC server i'm using is https://github.com/mpromonet/webrtc-streamer
on the streamer, i'm not seeing any connection attempts made either, is there a URL issue, or deviation from WebRTC?
let me know if there is anything else i can get to troubleshoot this.

icebal, 2021-03-02:
looking further, it looks like the appended CacheBust doesn't work and gets a 404 for this server:
this works:
http://octopi:8000/webrtcstreamer.html?video=printer
this does not:
http://octopi:8000/webrtcstreamer.html?video=printer&cacheBust=1614663038486
the cacheBust/gulp-cachebust NPM libs are also deprecated, so should be getting removed soon anyways.

cadriel, 2021-03-03:
Thanks for spotting this.

I can remove the cachebust param from the URL when the type is set to IP Camera - will get that in for next release.

Would appreciate if you could confirm if this fixes your issue once that happens.

Cheers!

cadriel, 2021-03-05:
1.10 should have fixed this. Please let me know! :)

icebal, 2021-03-06:
will update and test tomorrow after current print is done, thanks!

icebal, 2021-03-09:
IP camera still isnt showing up on the dashboard, but right clicking it, copy and pasting the Video Location to a new tab works still.
I also tried it with mjpg_streamer and the mjpg option, same issue, no video shown but link works in new tab.
Lemme know if you need any supporting info

cadriel, 2021-03-12:
So, the only thing the camera type drop down does - is swap the HTML tag between an <img> tag, and a <video> tag. With the img tag, the cachebust is applied - with the video tag, it is not.

Looking at the docs for WebRTC server, it looks like it'd require some specific implementation to support connecting to the stream - which is not something I've currently implemented.

I'm not sure if this is a common requirement for web rtc streams or not .. so if you have any thoughts or information on that, let me know.

shinuza, 2021-04-15:
@icebal I closing this for now, please comment if the issue persists

closed 2021-04-15 by shinuza