fluidd-core/fluidd #164: dashboard option to select multiple cameras

brushby, 2021-03-02:
I have two cameras on my printer --one to view the bed and another that views a closeup of the nozzle.

I would really like a way to quickly switch between the cameras on the dashboard. I know I can do that through the UI settings but that is a lot of steps when all I want to do is a take an occasional brief look at the other camera.

What I would like is to :

1. have the UI settings allow multiple cameras to be specified, possibly also allowing each to have a name.
2. have radio buttons at the top of the camera window on the dashboard in order to select which camera was being displayed.

It would be even better to be able to look at multiple camera windows at the same time but I would be very happy even if I could quickly select which camera I was using in a single camera window as described above.

thanks for considering this suggestion.

cadriel, 2021-03-03:
Yep, this sounds like a sane enhancement. No ETA as yet, but i'll add it to the TODO.

phorsfi, 2021-03-07:
Im stoked on this one too... just waiting on my endoscope nozzle camera

brushby, 2021-03-30:
You've added multiple cameras! Yay! Thanks very much --it makes things much easier for me.

closed 2021-03-30 by cadriel