fluidd-core/fluidd #166: Thumnail shows from previous print

markwinger, 2021-03-03:
When I print gcode that contains a thumbnail, the images shows up in the print panel of the dashboard.
But If I then print a gcode file that does not contain a thumbnail, the thumbnail of the previous print remains on the panel.

cadriel, 2021-03-03:
Ok, I *think* I have this figured out.

Please re-test and let me know if its still an issue with the next release.

Thanks for reporting!

markwinger, 2021-03-04:
I updated to the latest but the version and not change.  Update says v1.9.0.  I don’t think I have the fix.  How do I get it?

cadriel, 2021-03-04:
1.9 was released a while ago, this change has not been released yet.

cadriel, 2021-03-05:
1.10 is now out. Please check with this release! :)

markwinger, 2021-03-08:
I tested it on v1.10.  The problem appears to have been corrected.

cadriel, 2021-03-09:
Fantastic, thanks

closed 2021-03-09 by cadriel