fluidd-core/fluidd #167: Web page issue

Ducati87, 2021-03-04:
Hello guys, after I have installed of modules I get this errore when I try to load the web page:

`Traceback (most recent call last):
  File "/home/pi/moonraker-env/lib/python3.7/site-packages/tornado/web.py", line 1681, in _execute
    result = self.prepare()
  File "/home/pi/moonraker-env/lib/python3.7/site-packages/tornado/web.py", line 2430, in prepare
    raise HTTPError(self._status_code)
tornado.web.HTTPError: HTTP 404: Not Found`

this is the moonraker log:

`--------------------Moonraker Log Start--------------------
Git Version: v0.3.1-29-g63c685e
2021-03-04 08:56:54,738 [moonraker.py:add_log_rollover_item()] - 
#################### Moonraker Configuration ####################

[server]
host = 0.0.0.0
port = 7125
enable_debug_logging = True
config_path = /home/pi
klippy_uds_address = /tmp/klippy_uds

[authorization]
enabled = True
api_key_file = ~/.moonraker_api_key
trusted_clients = 
	127.0.0.1
	192.168.0.0/16
	::1/128
	FE80::/10
cors_domains = 
	http://*.local
	http://my.mainsail.xyz
	https://my.mainsail.xyz
	http://app.fluidd.xyz
	https://app.fluidd.xyz
	http://192.168.178.45
	http://192.168.178.45:*

[update_manager]

[update_manager client mainsail]
type = web
repo = meteyou/mainsail
path = ~/mainsail

[update_manager client fluidd]
type = web
repo = cadriel/fluidd
path = ~/fluidd

[system_args]
configfile = /home/pi/moonraker.conf
logfile = /tmp/moonraker.log
software_version = v0.3.1-29-g63c685e

#################################################################
2021-03-04 08:56:54,753 [authorization.py:__init__()] - Authorization Configuration Loaded
Auth Enabled: True
Trusted Clients:
127.0.0.1
192.168.0.0/16
::1/128
fe80::/10
CORS Domains:
http://.*\.local
http://my\.mainsail\.xyz
https://my\.mainsail\.xyz
http://app\.fluidd\.xyz
[... 1629 lines cut ...]
[stepper_x]
step_pin = PC2
dir_pin = PB9
enable_pin = !PC3
microsteps = 16
rotation_distance = 40
endstop_pin = ^PA5
position_endstop = 0
position_max = 220
homing_speed = 70

[stepper_y]
step_pin = PB8
dir_pin = PB7
enable_pin = !PC3
microsteps = 16
rotation_distance = 40
endstop_pin = ^PA6
position_endstop = 0
position_max = 220
homing_speed = 70

[stepper_z]
step_pin = PB6
dir_pin = !PB5

cadriel, 2021-03-05:
This is not a Fluidd issue, but that of a moonraker / klipper configuration issue.

I would note that it appears you're using Python 3, which is unsupported.

Your best bet would be to use a pre-compiled image like FluiddPI.

Otherwise, your best place for support with an issue like this is the Klipper discord.

closed 2021-03-05 by cadriel

knoopx, 2021-07-02:
hi @cadriel I'm also getting this error (404 on `/auth/oneshot_token`) on the latest version (v1.15.0). moonraker runs on py3, it actually doesn't run on py2. `oneshot_token` api is only available when `[authentication]` module is loaded, which is optional. Are we forced to use authentication now?

cadriel, 2021-07-02:
The `[authentication]` module is required. Having it doesn't necessarily require user logins tho.

knoopx, 2022-04-25:
adding this for future reference: you also need libsodium when enabling authorization otherwise will silently fail to initialize the module