fluidd-core/fluidd #170: Tachometer bug

Afterschnitzel, 2021-03-05:
Hello, I just updated Fluid to 1.10.0
I like rpm meter feature. But there are bugs.
Nozzle fan rpm are showing at part fan and part fan rpm are showing at nozzle fan.
And I really can live without 12 decimal numbers. Is possible to remove decimals and make digits bigger?

cadriel, 2021-03-05:
The extra digits i'll take care of.

However, the report re: rpm showing for different fans - if true, would be a klipper issue I think.. since I only report what is given to me for those values. Will double check my side tho.

cadriel, 2021-03-05:
A fix for this is in develop, it'll be in the next release.

I couldn't see any reason why rpm's would be swapped between fans..

closed 2021-03-05 by cadriel

Afterschnitzel, 2021-03-06:
Oh, sorry, cadriel.
It was my fault, I checked klipper config and I had to repair my fan pins. Now it is OK.
Thanks for your response.

[image]