fluidd-core/fluidd #178: Can't upload to Klipper through Prusa slicer anymore

Xenomorphdelombre, 2021-03-17:
Hi,

First of all, thanks for this amazing and easy to use interface for Klipper.

I've updated to version 1.10 and since then, I can no longer upload to "octoprint" in prusa slicer. I get this error message:

Error uploading to print host:
HTTP 404: Traceback (most recent call last):
File "/home/pi/moonraker-env/lib/python3.7/site-packages/tornado/web.py", line 1681, in _execute
result = self.prepare()
File "/home/pi/moonraker/moonraker/app.py", line 443, in prepare
raise tornado.web.HTTPError(404)
tornado.web.HTTPError: HTTP 404: Not Found

cadriel, 2021-03-17:
https://docs.fluidd.xyz/configuration/moonraker#octoprint_compat-block

closed 2021-03-17 by cadriel

Xenomorphdelombre, 2021-03-17:
Thank you! Good to know, that changed, it was enabled before the update. I've added [octoprint_compat] to ~/klipper_config/moonraker.conf