fluidd-core/fluidd #180: I don't understand your example Macros. Where does "BASE_RESUME" come from

anjoku, 2021-03-17:
Hi,
I don't understand a line in the example macros.

Where does the BASE_xxxxx entries come from.

BASE_PAUSE or BASE_RESUME

Example:

`[gcode_macro RESUME]
rename_existing: BASE_RESUME
default_parameter_E: 1      # edit to your preferred retract length
gcode:
    G91
    G1 E{E} F2100
    G90
    RESTORE_GCODE_STATE NAME=PAUSE_state MOVE=1
    BASE_RESUME`

in the example BASE_RESUME is renamed first and BASE_RESUME is listed again below.
What does that do?

Thank you :-)

Andre

cadriel, 2021-03-17:
The `rename_existing` prop allows you to rename and replace an existing, built in klipper macro with one of your own.

So - with the example above, we replace the `RESUME` macro with our own - and then also ensure we run the `BASE_RESUME` so that any built-in logic with the klipper resume still function.

closed 2021-03-17 by cadriel

anjoku, 2021-03-18:
Hi,

many thanks.

So the existing Klipper RESUME is renamed BASE_RESUME.

Then it will be executed as your "RESUME gcode".
And at the end again the internal Klipper Macro, which was renamed and still executed?

If so, why is the internal Klipper macro also being executed?

Best regards

Andre