fluidd-core/fluidd #181: Feature request: GCODE preview

Mechazawa, 2021-03-18:
A gcode preview like octoprint has would be very usefull, especially if it can sync up to the current print

blerrgh, 2021-04-06:
Seconded!

Mechazawa, 2021-04-18:
https://user-images.githubusercontent.com/1144171/115144667-2b57ce00-a04e-11eb-87f3-c18614fe2e9b.mp4

Mechazawa, 2021-04-19:
Just needs a bunch refactoring and UI changes and then I guess it's done

https://user-images.githubusercontent.com/1144171/115169943-b0c79680-a0bf-11eb-8644-67eeda572a61.mov

johngianni, 2021-04-20:
Very Nice !

Kragrathea, 2021-06-12:
Hi all. I am the author of the PrettyGCode plugin for OctoPrint. https://github.com/Kragrathea/OctoPrint-PrettyGCode

I recently decided to move to Fluidd and I would like to get my plugin working with it. Possibly in the UI or more likely as a stand alone web app that uses the API. 

To figure that out I would like to try out this GCode preview. But I can't figure out how to enable it.  I am not yet familiar with Vue development. I assume it would have to be done by rebuilding Fluidd?

StuSerious, 2021-06-14:
hey there @Kragrathea!
it's an honor to have you!
Currently @Mechazawa is very busy and his PR needs some of the final work. that's why it's currently disabled.

If you and a more effective back-and-forth communication, please join our discord here: https://discord.gg/GZ3D5tqfcF
We have internal testing branches that you could use and a lot of the devs are there to give you a hand!

Feel free to message me @ StuSerious#0404 so I can set you up with the proper roles :)
I wish you a great day!
~S

cadriel, 2021-06-14:
Hi @Kragrathea 

This sounds great. I'd really like to make the gcode viewer a first class citizen when it comes to Fluidd. As mentioned, we already have some progress on this - however, if you're willing to put in the effort to maintain and implement something I'd be more than willing to help with this integration in Fluidd.

As Stu mentioned, join us on Discord and we can discuss further! :)

Kragrathea, 2021-06-14:
Thanks for the quick replies! I did a bit more research and I think the first thing I need to do is get my code talking directly to Moonraker. After that I can figure out how to plug it into the Fluidd UI. So, I am not going to try to rebuild Fluidd with this GCode viewer right now.

maruhe, 2021-06-22:
i updated to current release today
gcode-viewer is nice, but here the 'follow progress' seems not to work propper
also 'show current layer' would be a great feature and auto zoom to object sice
'follow progress' disables it self a bit oftain
and about layout - i'd prefer the bar's to change layer and progress below the preview - currently they need a lot of space
[image]
looking forward to 3-column-dashboard :-*

arjanmels, 2021-09-11:
Very nice!

Only the follow progress is not in sync with actual printing: the total number of layers is different between the preview and klipper status and there is an offset of a couple of layers between the current layer. Also the within layer progress (which is the most difficult I guess, is off).

closed 2022-03-31 by pedrolamas

pedrolamas, 2022-05-06:
Thanks @harellevy, we might consider that for 1.19! 😉

PR #620 (2022-03-31): feat: graduates g-code previewer to non-beta
Closes #181