fluidd-core/fluidd #186: No moonraker connection. Please check moonraker status and / or refresh.

7effrey89, 2021-03-24:
I just updated moonraker through Fluidd, and now it seems moonraker wont load..

anyone got the same issue?

my version of moonraker, according below file:
/home/pi/moonraker/scripts/version.txt

env_version=0.2.6
deps_version=0.4.1

[image]

fredericarpin, 2021-03-25:
Same issue here, trying to start the moonraker from the command line also fails.

moonraker.log output follows:
`--------------------Moonraker Log Start--------------------
Git Version: v0.4.1-85-gf0a6620
2021-03-25 05:28:13,222 [moonraker.py:add_log_rollover_item()] - 
#################### Moonraker Configuration ####################

[server]
host = 0.0.0.0
port = 7125
enable_debug_logging = True
config_path = ~/klipper_config
temperature_store_size = 600
gcode_store_size = 1000

[authorization]
enabled = True
cors_domains = 
        *.local
        *.lan
        *://app.fluidd.xyz
        *://192.168.*
trusted_clients = 
        10.0.0.0/8
        127.0.0.0/8
        169.254.0.0/16
        172.16.0.0/12
        192.168.0.0/16
        FE80::/10
        ::1/128

[update_manager]

[update_manager client fluidd]
type = web
repo = cadriel/fluidd
path = ~/fluidd

[system_args]
configfile = /home/pi/klipper_config/moonraker.conf
logfile = /tmp/moonraker.log
software_version = v0.4.1-85-gf0a6620

#################################################################
2021-03-25 05:28:13,227 [moonraker.py:main()] - Moonraker Error
Traceback (most recent call last):
  File "/home/pi/moonraker/moonraker/moonraker.py", line 670, in main
    server = Server(system_args, file_logger)
  File "/home/pi/moonraker/moonraker/moonraker.py", line 74, in __init__
    self.moonraker_app = app = MoonrakerApp(config)
  File "/home/pi/moonraker/moonraker/app.py", line 89, in __init__
    self.auth = Authorization(config['authorization'])
  File "/home/pi/moonraker/moonraker/authorization.py", line 38, in __init__
    f"Unsafe CORS Domain '{domain}'.  Wildcards are not"
confighelper.ConfigError: Unsafe CORS Domain '*://192.168.*'.  Wildcards are not permitted in the top level domain.
2021-03-25 05:28:13,228 [moonraker.py:main()] - Server Shutdown
`

fredericarpin, 2021-03-25:
Looking at this commit: https://github.com/cadriel/FluiddPI/commit/6cbc934bee3b47699a61e179f1464cea18c1a25c#diff-ff2f27900f03eace50f784f6e3726bc66570e4dfbcca24d1ed5c3793e2e3a70c helped me figure out a way to potentially fix the problem. This worked on my pi, YMMV.

1. Connect to the pi via ssh
2. Make a copy of the `/home/pi/klipper_config/moonraker.conf` file
3. Edit the `/home/pi/klipper_config/moonraker.conf` and remove the `*://192.168.*` line under the `cors_domains:` section
4. Restart moonraker: `sudo systemctl restart moonraker`
5. Click the "Force Refresh" button on the webui

7effrey89, 2021-03-25:
@fredericarpin not sure if it's the way to go, but it worked!

closed 2021-03-25 by 7effrey89

elcojacobs, 2021-04-11:
Thanks @fredericarpin.
After updating to the latest version I got a bad gateway 502 on moonraker. Removing that line fixed it here too.