fluidd-core/fluidd #189: Feature Request: 3 Column Layout with Resizable Cards

11chrisadams11, 2021-03-25:
I think it would be nice to have an option to have a 3 column layout on the dashboard. Also, the ability to resize individual cards would be helpful. I would prefer to not have to scroll to see everything while printing, and currently some cards are just too large for my liking.

Thanks for the consideration

phorsfi, 2021-04-08:
I have a very WIDE display on my printer which runs Fluidd in a browser via the Pi and would love to have drag/drop cards as well as a selection for multiple columns... The display is 1920x515px, 12" wide by about 5" tall and it would be great to use fluidd more easilly!

maruhe, 2021-06-20:
3 Column Layout would be great - while center-column should be wider like OctoPrint with UI-Customizer

[image]

grantemsley, 2021-12-10:
I'd really love to see this as well. I'm using a 4k monitor and there's tons of space on the sides that I wish I could make use of.

XOIIO, 2022-05-20:
I'd also really like to have this, it seems like it should be simple. I have an 34" ultrawide monitor and fluidd really doesn't make great use of it. Seems like it should be pretty simple to add in for the layout editor too.

PR #719 (2022-07-04): feat: allows 4 layout columns in dashboard
[image]

[image]

Closes #189
closed 2022-07-04 by pedrolamas