fluidd-core/fluidd #191: Feature Request: Move print-time chooser to "Printing" dashboard widget

wrex, 2021-03-25:
I'm absolutely loving Fluidd. Really excellent work.

I've been experimenting with the different print-time choices (duration only, slicer, file estimation, filament). From a UI standpoint, it would sure be nice if the  Printing widget had a gear icon like the console widget that put up the dialog to make the choice.

It's just the tiniest of annoyances, of course, but having to click the three vertical dots, then UI settings, then find Print Times in the wall of various settings before clicking back to the dashboard requires a bit of mental context switching.

cadriel, 2021-03-27:
Yep, this seems like a viable request - i'll look into it for the next release.

cadriel, 2021-07-01:
The next release will remove the estimate settings, and move to displaying all estimates in one place. Closing this issue as I think it's sorted. Please feel free to comment here if the next release doesn't cover this off! :)

closed 2021-07-01 by cadriel

wrex, 2021-07-01:
Ooh. I was an evil PM overlord in my other life. I **LOVED IT** when the devs came up with a more elegant solution than the PMs spec'd in the feature request. It let me launch into the "spec the REQUIREMENTS dammit, let them figure out the best implementation" speech. 

All estimates in one place sounds great. :-)