fluidd-core/fluidd #200: Interface Language not Working: fluidd: v1.11.2-1c61777

PM3Delta, 2021-03-29:
Hi guys, hi cadriel

as of the latest update the interface language can not be set independently of the local device language.  E.g. local language in German but Interface Language set to English.  I tried rebooting etc.  Only if I set the local language to English as well, than the interface language is also in English.  Seems like the interface language setting is overwritten. Error affects PC, iPhone and iPad.
Thanks for looking into it.
BR
Peter
[image]

cadriel, 2021-03-30:
I'm not sure if this is still an issue with the 11.2 release or not, however - I've made some changes to allow setting the language back to browser default.

I've also tested this by setting chrome's preferred language to german, and ensuring that

a) By default, Fluidd selects german and;
b) When selecting a different language, this gets overridden correctly.

I can't test if this will behave similarly when the operating system's language is different, but I suspect it will.

Please re-open this if the issue persists after the next release, and thank you for reporting it! :)

PM3Delta, 2021-04-11:
Hi Craig,

sorry to come back to this.... unfortunately the bug still persists with the latest version v.1.12.2.

Could you please look at it again, something most have changed with v1.11.2, which caused this, as it was all good before.

Thanks a lot.

BR
Peter

cadriel, 2021-04-11:
Ok, re-opening. I'll try to figure this out for the next update.

cadriel, 2021-04-12:
We're having trouble replicating this. I have a couple of people, at least one of which has a different OS language that can change language fine, irrelevant of language set in your OS.

OS language should also not really have an effect here.

A couple of things to try;
1. Try setting the language to "Browser Default", then reloading. Then trying to set a new language again.
2. Stop the moonraker service, move the moonraker database to a temporary location - and then restart the service. This will cause moonraker to create a new database (and thus, new settings.. ). You can move the old database back after testing if you'd like to retain history etc.

If neither of the above options work, then we'll need to dig deeper and perhaps inspect errors in your console. If you already know how to do this, great! If not - perhaps a chat in Discord might be beneficial.

PM3Delta, 2021-04-13:
Tada ... success ... thank you so much for the hint.  No. 1 did it already. "setting the language to "Browser Default", then reloading. Then trying to set a new language again."

Sorry I didnt think of this myself and thanks a lot for looking into it despite it.

Topic can be closed.

[image]

cadriel, 2021-04-14:
Ok, good to hear! Closing! :)

closed 2021-04-14 by cadriel