fluidd-core/fluidd #208: Multi-Printers, interface mgmt error

kintarojs, 2021-03-30:
Environment:
fluidd v1.11.2
klipper v0.9.1-390
mainsail v1.4.0
moonraker v0.5.0-0
2 printers running on the same host.

When an update on interface settings was done, second printer take the same interface configuration:
Same name,
Same interface color,
Same macro filter

constantinef, 2021-03-30:
Hi, i have same issue https://github.com/cadriel/fluidd/issues/197
You need to add the following string to the moonraker.conf in server section for each printer:
database_path: ~/moonraker_database

example:
`[server]
host: 0.0.0.0
port: 7126
enable_debug_logging: False
config_path: ~/klipper-2_config
database_path: ~/moonraker2_database
klippy_uds_address: /tmp/klippy-2_uds
temperature_store_size: 600
gcode_store_size: 1000`

So that each printer stores its configuration in a separate database moonraker

constantinef, 2021-03-30:
I've reinstall fluiddpi, and that help me with a color theme