fluidd-core/fluidd #213: Feature Request: Adding the number of printed layers

SpeedyRS2, 2021-03-31:
I have a feature Request: Is it possible to add an additional information about the layer in the "printing" area?
Would be great, if I can see the number of the current Layer. For example: Layer xx from xxx
A good example is here on the mainsailOS screenshot:
https://github.com/meteyou/mainsail/blob/master/docs/assets/img/screenshot.png

Thank you :-)

cadriel, 2021-04-01:
Mainsail does not account for variable layer heights. In the case of a print with variable layer heights, Mainsail's reporting would be inaccurate - because moonraker currently does not provide the current layer.

Because of this, I chose to not add it because I have no reliable way to report ACTUAL layer count.

You can imlement by way of M117 and layer change gcode with your slicer tho. Fluidd will display the value of your M117 in the status panel, so you can report current layer via that.

closed 2021-04-01 by cadriel

seebaer1976, 2021-04-06:
you can also simply take the layer specification from the gcode e.g. this one:   ; Layer [layer_num]