fluidd-core/fluidd #215: Allow connecting to Moonraker using API key

W-Floyd, 2021-04-01:
While adding a Moonraker API host, would add and use an API key instead of relying on CORS or trusted IP. This would allow for secure external hosting using basic auth on a HTTPS reverse proxy of Fluidd, and API key only access to Moonraker. As is, however, it complains about CORS when it is not in fact a CORS issue.

cadriel, 2021-04-05:
I believe there are already people running Fluidd, externally - using reverse proxies and vpn gateways. Adding basic auth via nginx or some other method shouldn't be an issue.

If you're getting CORS errors from Moonraker, and you feel they are in error - it might be worth reporting this as an issue on the moonraker github.

Secondarily, Fluidd has no backend, other than moonraker - so from an auth point of view, there's no secure way to maintain an API key for the purposes of your example, yet.

This IS something I'd like to resolve tho, but I'm not quite there yet.

cadriel, 2021-04-06:
I'm going to close this for now. Better auth is coming.

closed 2021-04-06 by cadriel