fluidd-core/fluidd #222: Themes not applied correctly between multiple printers

OhDeeEsTee, 2021-04-03:
Hi Cadriel. Loving Fluidd so far, excellent work. I'm having an issue where my themes are not being applied correctly. I'm running 3 printers through the app.fluidd.xyz interface server. Whenever I change printers the set theme from the previous printer is applied to the one I'm switching to. This occurs for all elements except the main Fluidd logo which remains correct as does the theme setting in the ui options.

cadriel, 2021-04-04:
This sounds like the moonraker_db issue.

Are your printers all configured on one pi? If so, you need to configure each moonraker instance to use its own unique database - because, by default - it uses the same one.

I otherwise can't reproduce this if its something else.

You can find the docs for moonraker here;
https://moonraker.readthedocs.io/en/latest/configuration/

Specifically look out for the `database_path` option.

OhDeeEsTee, 2021-04-04:
No, 3 printers, 3 pi's, 3 separate installs, but I'll still look into the moonraker db

closed 2021-04-04 by OhDeeEsTee