fluidd-core/fluidd #225: Bed Mesh graph view scaled incorrectly making it seem as it the X and Y axes are swapped

tehbiga, 2021-04-06:
***Scenario:***
Bed Mesh leveling with a rectangular pattern instead of square.

***Problem:***
With a rectangular bed the mesh leveling graph seems to show the axes incorrectly _(wider bed shows as deeper in bed mesh graph),_ but when mousing over to check the positions and height the reported values seem correct.

https://gfycat.com/UnsteadyBestJackal _(HD version is legible)_

***Assumption:***
I assume this is just a display issue with Plotly or how it's configured and not an actual data consumption problem.

***Versions:***
_Browser:_ `Firefox 87`
_Fluidd:_ `v1.11.2`
_Moonraker:_ `v0.5.0-6`
_Klipper:_ `v0.9.1-399`

***Most relevant config:***

```[bed_mesh]
speed: 40
horizontal_move_z: 5
mesh_min: 20, 20
mesh_max: 190, 170
probe_count: 5, 5
relative_reference_index: 12
```

cadriel, 2021-04-07:
The next release has a new mesh visualizer, and I handle the mesh a bit differently. I've done a quick test this side to confirm. The screenshot below, I reduced the dimension of my Y axis, and this seems to present properly.

Once the new release is out, let me know if you still encounter issues.

[image]

Thanks for the report!

closed 2021-04-07 by cadriel

BillyNate, 2023-09-21:
This might need some further investigation.
I'm on Fluidd 1.25.3, and with a rectangular bed it doesn't visually meet the real world dimensions.

Here's an image of both the current situation and what I would expect it to look like, making it easier to interpret the mesh visualization:
[image]

Setting the smaller axis min/max to the same length as the longer axis will result in the adjusted mesh 😉