fluidd-core/fluidd #228: Feature Request: identify pi issues (undervoltage / overheating) via get_throttled

cfelicio, 2021-04-08:
Hi There,

Is it possible to detect / display any issues with the raspberry pi? Similar to this:

[image]

Reference: https://www.raspberrypi.org/documentation/raspbian/applications/vcgencmd.md

Thanks! :-)

cadriel, 2021-04-08:
This is something I can't detect at the moment.. but will come as soon as its available.

cadriel, 2021-04-12:
Ok, so throttle detection is in place. I think I can start looking at solutions here.

cadriel, 2021-05-07:
This is in place with the latest release.

Very much looking for feedback on current implementation, so feel free to comment here or join the Discord!

Thanks!

https://discord.gg/GZ3D5tqfcF

closed 2021-05-07 by cadriel

cfelicio, 2021-05-08:
Looks awesome, thanks so much! Just tried with a crappy power supply, got those right away:

[image]

intensite, 2021-05-12:
Ok I have the same messages but can't get rid of them.
The clear_all button has no effect.

cfelicio, 2021-05-12:
Did you try rebooting the Pi? For me the message would not go away with the weak PSU either, but I assumed its because the problem was still happening. I did not test against an intermittent scenario.

intensite, 2021-05-12:
Unfortunately, rebooting doesn't seem to have any effect

I consider the power supply is good as it provides 2.8Amps at 5v.  (I will however try another one)
I also have a Fan to cool the enclosure where the RasPi and printer board is encased.   

I looked-up the [RaspberryPi  documentation here](https://www.raspberrypi.org/documentation/raspbian/applications/vcgencmd.md)

When I run the command:
```
pi@tarantula:~ $ vcgencmd get_throttled
throttled=0x50000
```
0x50000 = 0101 0000 0000 0000 0000
From this we can see that bits 18 and 16 are set, indicating that the Pi has previously been throttled due to under-voltage, but is not currently throttled for any reason.

I also ran the following:
```
pi@tarantula:~ $ vcgencmd get_throttled
throttled=0x50000
pi@tarantula:~ $ vcgencmd measure_volts
volt=1.2000V
pi@tarantula:~ $ vcgencmd measure_temp
temp=42.9'C
```

According to the docs, everything is norminal but these flags apparently cannot be reset :-(  
So it is not a Fluidd issue.

From [RaspberryPi Forums Link](https://www.raspberrypi.org/forums/viewtopic.php?t=241736) the bits have the means of:

Bit Meaning
0 Under-voltage detected
1 Arm frequency capped
2 Currently throttled
3 Soft temperature limit active
16 Under-voltage has occurred
17 Arm frequency capped has occurred
18 Throttling has occurred
19 Soft temperature limit has occurred

Maybe we could just use bit 0 and 2 and report it is 'currently throttled' or not and bit 18 for 'throttled has occured'

intensite, 2021-05-12:
Ok,  I think I have found the culprit.

If I restart the RasPi with the printer's board disconnected from the Pi's USB port, the problem seems to go away, I get the following:
```
pi@tarantula:~ $ vcgencmd get_throttled
throttled=0x0
```

So the power supply is fine for the Pi itself but causes a brownout when it starts the printer's board.  
I have been using it like that for many months without any problems, only the latest version of Fluidd brought this issue to daylight.  So I am not too concerned.  Only annoyed at the now permanent warning :-)

I recall reading a wile ago about some hacks to prevent the RaspPi from powering the printer's board https://community.octoprint.org/t/put-tape-on-the-5v-pin-why-and-how/13574
I might try it.

cadriel, 2021-05-13:
Note, the official pi supply for a 4 is 5.1 volts.

Also, the reason the 'previously..' notifications stick is because they're constantly active until a host reboot. You can't clear'em because they're always there.

If they re-appear on a reboot, then something is triggering them again.

intensite, 2021-05-13:
Yep that's what I have learned from the official docs as mentioned above.

The issue I have is, when the system boots, the power drawn from the RasPi 3B,  plus the MKS Gen printer board, plus the USB camera,  momentarily cause a "brownout" condition which is what is reported by the Pi Throttling registers.  

Once stabilized everything is OK, but as you said, the registers are reset only at boot time.  

Once the raspberry Pi is up and running, I have a macro to energize a relay to turn on the main power supply for the printer (which also power the printer board) so the power is not an issue after that.  

A can live with the small annoyance of the warning for now until I find a better power solution. 
[image]

As I said it is not a bug in Fluidd. 

Thanks for this wonderful interface.

guangxb, 2021-06-15:
https://retropie.org.uk/forum/topic/1006/under-voltage-on-the-pi3/8

cadriel, 2021-06-15:
This was from 2017 - I believe we're on Kernel 5 now. :)

iPeel, 2021-07-14:
Hi, is there any way to disable these messages and the popup? I can't see any config option in settings.

Thanks.

StuSerious, 2021-07-14:
hey there @iPeel!

there is no way to disable warning popups, they appear permanently and are a direct UI representation of `vcgencmd`.
I would advise to look into resolving the problem, since it can have varying consequences on your experience; from degrading your SD to crashing prints!

feel free to [join our discord](https://discord.gg/GZ3D5tqfcF) for support! 
cheers!
~S

iPeel, 2021-07-14:
There's no underlying issue, it's just we run RPi 3a's on a print farm and usually only ever go to the UI to do admin stuff, so regular prints are always perfect but whenever you do anything like restart a service or update you end up having to accept a popup.

It's fine though, I'll just keep clicking!

abedaltahrawi, 2022-01-16:
just added avoid_warnings=2 to /boot/config.txt and the msg gone