fluidd-core/fluidd #235: Error uploading gcode file

stpavel, 2021-04-09:
When I try upload file , I have get error after process uploading 
**Request failed with the status code 405**
At the same time, I can delete or create directories from the file panel without any problems, that is, everything is ok with permissions

cadriel, 2021-04-12:
405 is method not allowed, this feels like some other issue.

Do you have any other warnings? Are you connected to klipper? How did you install Fluidd?

stpavel, 2021-04-12:
No any other warinings . Yes, I connected to klipper. Install trough kiauh script

cadriel, 2021-04-12:
Are you uploading via button, or drag and drop? Do you get a different result if you drag and drop?

There's an existing bug whereby using the upload button fails, but dragging and dropping works - so i'd like to rule that out.

stpavel, 2021-04-13:
I tried via button and drag 'n' drop, result the same .

cadriel, 2021-04-13:
I think you need to inspect your moonraker config. A 405 is being returned from it, so there's likely something else going on.

shinuza, 2021-04-14:
Could also be coming from nginx or another webserver in-between.

cadriel, 2021-04-18:
I'm going to close this - seems to be related / unique to user environment.

closed 2021-04-18 by cadriel