fluidd-core/fluidd #240: Feature request: Mainsail-like "Printing" widget

ssorgatem, 2021-04-12:
The one thing I like better from Mainsail compared to Fluidd is the "Printing" widget, because, compared to Fluidd:

- It shows all 3 remaining time estimations at the same time.  Very useful since, at least for me, they over and underestime each very differently.
- Bigger preview image. It just looks nicer. With Fluidd sometimes it's hard to tell the object in the thumbnail.
- Layer progress. Very handy in case the print is interrupted for some reason. Also, nice,

It would be nice to be able to optionally add these features to Fluidd's "printing" widget.

cadriel, 2021-04-12:
Yep. I've been meaning to give the status panel a once over now it's been about for a while.

I'd love to hear suggestions from other members of the community on what you'd like to see here too.

StuSerious, 2021-04-12:
I think what @ssorgatem was outlining is about what's needed
the one thing I miss about octoprint is the load/print procedure.

StuSerious, 2021-04-12:
@cadriel here are a couple of ideas on top of what @ssorgatem was suggesting:
- I find that the only thing I'm craving since I left octoprint (other than Spool Manager and Dry Run) is the Load/Print routine. being able to load a print, take a look at parameters and then hit the button to start it,
- an upload or print button on the top bar, to make it easy to have a file uploaded and ready with only one click after entering the UI
- the possibility to enable a sanity check / confirmation pop-up with something like "Is the Build Plate clear?"

hofleverancier, 2021-05-23:
> Yep. I've been meaning to give the status panel a once over now it's been about for a while.
> 
> I'd love to hear suggestions from other members of the community on what you'd like to see here too.

I'd like the option of showing the ETA next to or instead of time remaining.

cadriel, 2021-07-01:
Most items discussed here are coming up in the next release, so closing this.

closed 2021-07-01 by cadriel