fluidd-core/fluidd #241: z offset really wants four values

wrex, 2021-04-12:
## Affected Version(s):
(The versions of fluidd this issue occurs on.)
- klipper version: v0.9.1-417
- moonraker version: 0.5.0-11
- fluidd version:  v1.12.2

## Platform:
(Information about your platform, like the OS or browser you are using.)
MacOS 11.2.3
Chrome Version 89.0.4389.114 (Official Build) (x86_64) 

## Hardware Configuration:
(Information about your hardware configuration. Like MCU, host and setup)
Stock Voron V0

## Reproduction Steps:
1. Remove 0.001mm and 0.005 entries from UI preferences in Z Adjust values by clicking the trashcan icons next to their entries
2. restart the firmware or close and refresh the browser

## Actual Results:
Initially shows just the desired two entries (0.01 and 0.05) but after restart it repeats the entries so you have 0.01 0.05 0.01 0.05

## Screenshot(s):
Attached.

## Expected Results:
Should only have two entries

## Log Files:
(The log files fluidd provides at Configure > System Control.)
- klipper log:
- moonraker log:

[moonraker (1).log](https://github.com/cadriel/fluidd/files/6300280/moonraker.1.log)
[klippy (2).log](https://github.com/cadriel/fluidd/files/6300281/klippy.2.log)
[image]

StuSerious, 2021-04-12:
hey there @wrex, thank you for your detailed Bug Report!

I can replicate your issue with the reproduction steps you listed.
[image]

more updates to follow!

cadriel, 2021-04-13:
Yep, thanks for reporting. I've fixed this in develop, will be in next release.

closed 2021-04-13 by cadriel