fluidd-core/fluidd #248: 1.12.2  - Thumbnails not updated on New Upload - same filename

weasel0x00, 2021-04-15:
## Affected Version(s):
(The versions of fluidd this issue occurs on.)
- klipper version: v0.9.1-421
- moonraker version: v0.5.0-12
- fluidd version: 1.12.2

## Platform:
Accessing with MacOS using Safari and/or Brave(Chrome)

## Hardware Configuration:
This happens on my v0, v1, and v2, all running on PIs, FluiddOS or RaspberryPIOS

## Reproduction Steps:
1. Uploaded a GCode file through API in SuperSlicer
2. Edit the same setup, move, add, delete STLS, etc. and upload again with the exact same file name

## Actual Results:
The Thumbnail associate with that file is not updated, and still shows the one from the original / initial upload.

## Expected Results:
Thumbnail should be updated on file upload, even if a file with the same name already exists.

## Log Files:
Not sure these are relevant?  I can get them I you need though.

StuSerious, 2021-04-16:
hey there @weasel0x00, thank you for your detailed bug report, the issue is currently being looked at!

updates will follow soon!

cadriel, 2021-04-16:
The fix for this has turned out to be a couple of things.

- Moonraker required a change to remove the cache headers, along with no longer returning a 304.
- Fluidd needed to add a cachebust to the thumb image request.

Moonraker has deployed its change, Fluidd's change will be in the next release.

Note, there maybe other changes down the line to resolve cache issues to do with these thumbs - as currently, its not ideal to forcefully request a fresh image on each load.

Thanks for spotting this. I'm closing for now, but feel free to re-open if its still an issue after release.

closed 2021-04-16 by cadriel