fluidd-core/fluidd #249: Palette 2 Integration

cfelicio, 2021-04-16:
This is very fresh off the press, but klipper now officially supports the Palette 2 via connected mode: 

https://github.com/KevinOConnor/klipper/pull/4057

Before this feature, you would need to use Octoprint plugins to talk to the Palette, so it was pretty much mandatory to use Octoprint if you had a Palette.

The feature works really well, and I'm wondering if it's possible at some point to have the info the Palette generates on the Fluidd interface? Similar to what is currently displayed on the Octoprint plugin:

[image]

Appreciate your input!

cadriel, 2021-04-19:
Yep, I think this is valid. Its on the roadmap, no timeframe as yet.

chippawah, 2021-04-27:
@cfelicio Are you currently able to use the palette with Fluidd and it just doesn't show the output from the palette? I'm looking to replace Octoprint with Fluidd when I setup my Palette but wonder if I should stick with Octo until this issue is resolved.

cfelicio, 2021-04-27:
@chippawah Palette works fine with Fluidd as long as you configure it on the klipper.cfg. I can still see the pings on the Palette itself, so this is just a "nice to have".

ETE-Design, 2024-07-03:
Will this ever be implemented?

pedrolamas, 2024-07-04:
The reality is that this is not a core priority for Fluidd.

I personally don't have the hardware to test this so that makes things even harder, but if someone on the community wants to provide a code contribution for this (and ensure that either they or we can maintain this feature) we might take a look and see how we can integrate it.