fluidd-core/fluidd #252: Feature Request: Power Button on top

dushyantahuja, 2021-04-17:
Is it possible to add the power button for the printer at the header?

Preferably where the blue circle is in the image below. 
[image]

cadriel, 2021-04-18:
There's no real way to distinguish a power button, versus powering anything else. Also, power is a moonraker plugin - so it really needs its own unique place to manage because not every user configures this the same as others.

I'd also like to keep the header as minimalist as possible.

For these reasons I think adding a button up here is not doable.

Thank you!

closed 2021-04-18 by cadriel

dushyantahuja, 2021-04-18:
Sure - I was thinking more on the lines of - if the switch is called "printer" put it on the header. It would make things so much more convenient.

1ac3124, 2021-04-20:
@cadriel what do you think about power widget on dashboard like this?
[image]

cadriel, 2021-04-20:
Fluidd used to have the power widget on the dashboard in this form, but it takes up too much space for very little benefit, so was moved to the hamburger menu.