fluidd-core/fluidd #253: Feature Dashboard Sensors GPIO - (RaspberryPi / OrangePi)

PatricLeal, 2021-04-19:
## IIs your feature request related to a problem?
Currently, it is not possible to validate the operating history during a print or current status of a printer's resources, since we are only able to monitor the standard temperature sensors and temperature sensors of the Raspberry and MCU.

## Describe the solution you'd like:
The solution is very simple, with the use of a dashboard panel it would allow adding more sensors referenced to gpio or hardware, this way the current and historical information would be all on the screen, the solution is similar to the one used in PRTG for control  of iot devices or systems and servers, with the solution it would be possible to find situations in which it could be causing some failure or malfunction of equipment or malfunctioning situation.

## Describe alternatives you've considered:
There are monitoring solutions PRTG, TeamViewer iot, zabbix, nagios, however they are solutions that need other installed applications and hosting of servers or contracting third party services, in addition to losing in the integration factor which would be outside the current platform.

## Affected users and/or setups:
Change affects a group of users who want greater control and security on their printers, especially those using platforms with GPIO availability (RaspberryPi and OrangePi). The benefits are mainly the control of equipment reliability, control of electrical circuits, in addition to the possibility of  control external variables such as humidity, also allowing to turn off equipment or prevent printing in adverse or risky situations (for example, high humility of the environment, over voltage and power problems), such panel availability opens up new opportunities in the management of connected equipment.
----------------------
Prtg Example;
[image]

(Mainsail/Fluidd) Current temperature panel;
[image]

[image]

cadriel, 2021-04-20:
Most of this is outside of the intent of a printer UI, so likely isn't going to happen.

I will say that some data, like CPU throttling and under voltage warnings, along with CPU utilization and low memory warnings ARE coming, and may cover some of what you've mentioned above.

Thanks!

closed 2021-04-20 by cadriel

PatricLeal, 2021-04-21:
> Most of this is outside of the intent of a printer UI, so likely isn't going to happen.
> 
> I will say that some data, like CPU throttling and under voltage warnings, along with CPU utilization and low memory warnings ARE coming, and may cover some of what you've mentioned above.
> 
> Thanks!

Thank you, I will wait for this news, thanks for the return.