fluidd-core/fluidd #256: Fluidd confuses settings when multiple printers are configured

anarchocactus, 2021-04-20:
## Affected Version(s):
(The versions of fluidd this issue occurs on.)
- klipper version: v0.9.1-427
- moonraker version: v0.5.0-18
- fluidd version: v1.12.2

## Platform:
Safari 14.0.3 on MacOS X 10.15.7

## Hardware Configuration:
Printer: FLSUN Q5
Printer: TwoTribes Sapphire Pro

## Reproduction Steps:
Example with printer name:

1. Name first printer "Sapphire" in settings
2. Goto second printer and set name to 'Q5" in settings -> In the Printers List, both are now correctly named "Sapphire" and "Q5"
4. Now switch over to "Sapphire" -> Name is now "Q5" 

## Actual Results:
Name of first printer was apparently overwritten by naming second printer. Both have now the name "Q5". This is very confusing when working with more than one printer.

## Screenshot(s):

Step 1:
[image]

Step 2:
[image]

Step 3:
[image]

## Expected Results:
Printers should keep their names

## Log Files:
(The log files fluidd provides at Configure > System Control.)
- klipper log: [klippy-1.log.gz](https://github.com/cadriel/fluidd/files/6346164/klippy-1.log.gz)
- moonraker log: [moonraker-1.log.gz](https://github.com/cadriel/fluidd/files/6346166/moonraker-1.log.gz)

## Additional information
Furthermore the Jobs List in the Dashboard is also the same between the two printers, although they have different virtual_sdcard paths configured. Same for the dedicated Jobs list. Theming appears to have the same problem

cadriel, 2021-04-20:
Are both of your printers running from the same Pi?

cadriel, 2021-04-20:
Looking at your screenshot, this appears to be true.

The issue revolves around your moonraker configuration sharing a database. Please update your moonraker configuration to use a different database per instance;

See the `database_path` option here; https://moonraker.readthedocs.io/en/latest/configuration/

This is not a Fluidd issue. :)

closed 2021-04-20 by cadriel

anarchocactus, 2021-04-21:
Hi, Thanks a lot for the clarification. I use a KIAUH multi-instance installation. Possibly it is a small bug in their way of seperating the instances, as I cannot remember to have ever touched the moonraker configuration.