fluidd-core/fluidd #263: Possibility to lock all controls.

hofleverancier, 2021-04-26:
## Is your feature request related to a problem? Please describe.

Fluidd is amazing but I am always a bit uneasy with having a window open because a miss click can cause you to change a setting that will immediately (potentially catastrophically) affect a print in progress.

Btw I have found that especially on a phone it is extremely easy to accidentally move a slider.

Issue is somewhat related to issue #250, except I'm referring to all controls.

## Describe the solution you'd like:
A button at the top of the page that locks all controls, when unlocking a confirmation dialog shows up.

[image]

StuSerious, 2021-04-26:
hey there @hofleverancier!

we are seeing more and more reports about this here and on Discord, so Cadriel is already hard at work at making things safer!
you can take a look at the commit history to get an idea of how it's currently implemented.

the feature will be out soon™️ so in a bit you will be able to test it and see if it suits what you wanted!
feel free to join the Discord for more back and forth!
thanks for the FR 😃 

have a nice day!!
~Stu

cadriel, 2021-05-07:
I'm going to close this.

The latest release specifically addresses the slider controls on a mobile layout. Overall, I'd like to avoid a full lock if possible. Please feel free to comment again here, or join the Discord should you have further thoughts on this tho.

Thanks

https://discord.gg/GZ3D5tqfcF

closed 2021-05-07 by cadriel