fluidd-core/fluidd #265: recent update caused an Issue

csiller88, 2021-05-01:
I did the most recent updates and no receive:
"
Command format mismatch: endstop_state oid=%c homing=%c next_clock=%u pin_value=%c vs endstop_state oid=%c homing=%c pin_value=%c
This type of error is frequently caused by running an older
version of the firmware on the micro-controller (fix by
recompiling and flashing the firmware).
Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.
Protocol error connecting to printer
"

## Affected Version(s):
(The versions of fluidd this issue occurs on.)
- klipper version: v0.9.1-468
- moonraker version: v0.6.0-18
- fluidd version: v1.12.2

## Platform:
Chrome
(Information about your platform, like the OS or browser you are using.)

## Hardware Configuration:
(Information about your hardware configuration. Like MCU, host and setup)

## Reproduction Steps:
I updated moonraker.
Then I updated klipper.

## Actual Results:
(What happens after the reproduction steps have been followed.)

## Expected Results:
(What *should* happen after the reproduction steps have been followed.)

I expected it to just be a standard update. I can always flash the PI, but I wanted to see if this was a common issue.

## Log Files:
(The log files fluidd provides at Configure > System Control.)
- klipper log:
- moonraker log:

[moonraker.log](https://github.com/cadriel/fluidd/files/6409940/moonraker.log)
[klippy (1).log](https://github.com/cadriel/fluidd/files/6409941/klippy.1.log)

StuSerious, 2021-05-01:
hey there @csiller88! 
your issue is due to a recent commit in Klipper src files, that requires a re-flash.

therefore, your options to resolve are:
1. Reflash your MCU following your [board's directions](https://www.klipper3d.org/Installation.html#building-and-flashing-the-micro-controller);

2. Revert to the latest stable commit by running the following commands:
```sh
sudo systemctl stop klipper
cd ~/klipper && git reset --hard d1946fb
sudo systemctl start klipper
```

3. If you are a KIAUH user, use the 4) [Advanced] menu and perform a 2) [Rollback]

please do note that reverting is a temporary fix - we recommend updating your MCU.

reach out on [our discord](https://discord.gg/GZ3D5tqfcF) if you have further questions!
~S

Booner99, 2021-05-01:
I too have this problem. I have re-compiled the firmware but same problem exists.

cadriel, 2021-05-01:
@Booner99 @csiller88 

As per the messaging, you need to re-flash your MCU's. This isn't a Fluidd issue. For further support, feel free to reach out in the Fluidd discord, or the klipper discord.

Thanks!

closed 2021-05-01 by cadriel