fluidd-core/fluidd #266: Cannot update/recover klipper and moonraker

Smirzor, 2021-05-02:
## Affected Version(s):
(The versions of fluidd this issue occurs on.)
- klipper version: ??
- moonraker version: ?? 
- fluidd version: v1.12.2

## Platform:
(Information about your platform, like the OS or browser you are using.)
win10, edge chromium

## Hardware Configuration:
(Information about your hardware configuration. Like MCU, host and setup)
ender 3, skr mini e3 v2, raspberry pi 3

## Reproduction Steps:
cannot update klipper or moonraker and also recover won't work

## Actual Results:
error: "unable to recover" when i try to press the recover button

## Screenshot(s):
https://drive.google.com/file/d/1tpOX6s4bSOpah6qxY8HYAhCv9XWwEUQ3/view?usp=sharing

## Expected Results:
i should be able to update or recover, sadly it doesnt

## Log Files:
(The log files fluidd provides at Configure > System Control.)
- klipper log: https://drive.google.com/file/d/1HHaqI0lH_y3AgzIlhNqRJENYclboCWhu/view?usp=sharing
- moonraker log: https://drive.google.com/file/d/1TJ8QbIe18N_lK17OeUamsv2TivXepZEX/view?usp=sharing

StuSerious, 2021-05-02:
hey there @Smirzor 
this could be a duplicate of #265, have you tried the steps listed there?

thank you,
~S

cadriel, 2021-05-02:
Hi! GitHub issues are for tracking bugs and / or features. For support, or problems otherwise outside of Fluidds direct control, please see the various Discord servers.

Thanks!

closed 2021-05-02 by cadriel