fluidd-core/fluidd #268: Swap X and Y control

TazerReloaded, 2021-05-02:
## Is your feature request related to a problem? Please describe.
I have a camera mounted on the right side of the printer, and from the perspective of that camera, the X and Y axis are swapped (X -> forwards/backwards, Y -> left/right).

## Describe the solution you'd like:
An option to swap the controls for X and Y, just like the invert options. I use the manual controls to position a laser, which requires many small adjustments, would be easier if the controls were swapped.

## Describe alternatives you've considered:
Anything that involves changes in Klipper has too much impact on the printers behavior, it's just a control thing.

## Affected users and/or setups:
Users who have a camera on either side of their printer instead of behind or in front.

## Additional Context:
[image]

cadriel, 2021-05-02:
I don't think swapping controls specifically to meet the needs of a camera setup is the best option here. Secondarily, the invert buttons seemingly do the same thing.

Thanks!

closed 2021-05-02 by cadriel

TazerReloaded, 2021-05-04:
Well, the invert buttons invert axes (180deg), but my problem is that my controls are shifted 90deg, so X is Y (depth) and Y is X (width), because of the camera position. I'd like to swap the two axes so that the controls are more logical, so for example instead of `G0 X+10` the button sends `G0 Y+10` and vice versa when the option is toggled on.
Anyways, features are your decision, great UI, thanks for the hard work!

StuSerious, 2021-05-04:
I imagine you have a reason otherwise you wouldn't have opened an issue, but I'm still going to ask out of curiosity 😄 
why can't you move the camera? 🤔

TazerReloaded, 2021-05-04:
There is a glass window in the front, so no room there, and the printer touches the back of the enclosure, so no room there either. There is a second camera in the back right corner, but with that one it's even harder to move the head because its field of view is parallel to the print bed to monitor warping.