fluidd-core/fluidd #272: Fluidd on Pi3B+ hangs from time to time, print continues

vgroenewold, 2021-05-05:
This seems like a buffer issue, but I often (few times a day) have the problem that I execute something on the Fluidd UI (like bed leveling, or adjusting Z while printing, or... seems rather random) and the UI becomes unresponsive. I can still click around, but commands are not being send to the Pi nor received, prints always continue so my guess is that the Pi becomes too busy. But how can I diagnose that or even solve it?

Thanks

wlhlm, 2021-05-06:
The htop session freezing (but not the printer) indicates to me that It could be a poor network connection. If you are using WIFI you can try to use cable and see if it improves things.

cadriel, 2021-05-07:
Yep, I've seen this with a poor wifi connection too.

Suggest plugging in via ethernet and see if that resolves your issues. Also, make sure you've disabled sleep for your wifi adapter, this is a part of the latest fluiddpi releases - but if you're on an older image then it wouldn't have applied. You can do this by editing the `/etc/rc.local` file and adding this to the bottom of it;

```
# Disabling power management for wlan0
iw dev wlan0 set power_save off
```

closed 2021-05-07 by cadriel