fluidd-core/fluidd #273: Can the Emergency Stop button be moved to the left hand side of the page, yet right of the logo?

ayerlock, 2021-05-05:
## Is your feature request related to a problem? Please describe.
The current location of the Emergency Stop button has been causing issues with usage from both my tablet and phone.  Due to the nature of how individual touches work in Android, in the current location it is far too easy to accidentally hit it when pressing either the 3Dots or the Configure buttons.  This has caused me to accidentally cancel four prints now, one of which was an 8 hour print...just an hour shy of being done.

## Describe the solution you'd like:
Can the button be moved to the left side of the menu bar but maybe a half inch to an inch right of the logo.  This would still allow for easy access to it while reducing or even removing the potential for accidentally clicking it when using a tablet or phone interface.

## Describe alternatives you've considered:
There is currently no way to disable the button entirely (which I would do) so there really is no alternative.

## Affected users and/or setups:
Anyone who uses a tablet or phone to use the web interface.

## Additional Context:
### Current Location:  ###  
[image]  
  
### Proposed Location:  ###  
[image]

StuSerious, 2021-05-06:
hey there @ayerlock!
if you look around for a bit you'll see that this issue has been brought up more and more recently..

there is currently a change in develop to move the eStop button to a safer location on the screen. (a floating button on the bottom left of the UI)
here is a pic of it:
[image]

_personal opinion:_ there shouldn't be a way to disable it, eStop is a very important safety feature that needs to be quickly accessible by users, so maybe a tap and hold would be better..

ttyl
-S

cadriel, 2021-05-07:
Thanks for your thoughts on this. The latest release addresses this ticket, so I'm closing for now.

Feel free to join the discord or comment here if you've got further comments on the new functionality!

https://discord.gg/GZ3D5tqfcF

closed 2021-05-07 by cadriel