fluidd-core/fluidd #274: Printing wrong filel

markwinger, 2021-05-05:
## Affected Version(s):
(The versions of fluidd this issue occurs on.)
- klipper version:v0.9.1-480
- moonraker version: v0.6.0-22
- fluidd version: v1.12.2

## Platform:
Raspberry pi 3

## Hardware Configuration:
(Information about your hardware configuration. Like MCU, host and setup)
Ender 3, 
creality v4.2.7 mb, 
bltouch

## Reproduction Steps:
1. Selected a file with download and print.
2. A different object with the same file name already existed
## Actual Results:
(What happens after the reproduction steps have been followed.)
Printed the old file, but the thumbnail showed the new file.
## Screenshot(s):
(Images or gifs showing the problem, perhaps before/after.) 

## Expected Results:
(What *should* happen after the reproduction steps have been followed.)
Expected the newly downloaded file to print.  
Deleting the old file solved the problem.

## Log Files:
(The log files fluidd provides at Configure > System Control.)
[moonraker.log](https://github.com/cadriel/fluidd/files/6429127/moonraker.log)
[klippy.log](https://github.com/cadriel/fluidd/files/6429128/klippy.log)

- klipper log:
- moonraker log:

## Additional information
(Extra information relevant to the issue, links to similar issues.)

Arksine, 2021-05-07:
Which file was it?  Generally speaking this shouldn't be possible if the thumbnail was embedded in the gcode file, as the old file would have to have been overwritten to extract the new thumbnail.

With UFP files the gcode file is extracted and moved before the thumbnail, so if there is some kind of filesystem error overwriting the old gcode the new thumbnail should not be extracted.

Arksine, 2021-05-08:
If the file showed up in the list 2 times then it suggests that there was some difference in the file name.

markwinger, 2021-05-08:
I think the file name was CE3_mount.gcode.  I kept doing upload/print and it kept printing something else.  The I sorted by name and the same file came up twice, so I deleted the old one.  I uploaded and printed again then it worked.

markwinger, 2021-05-08:
I'm unclear about how this works.  When I was working with klipper on octoprint using virtual sd card I saw some anomilies I did not understand that looked something like this.  It was like there was 2 memory locations, real and virtual so you could have the file twice.  Can that happen here?

Arksine, 2021-05-08:
I looked closely at the log and see what happened.  You had two different files in your gcodes directory:
- CE2_mount.gcode
- CE2_Mount.gcode

The change in case sensitivity is what got you.  Klipper's `virtual_sdcard` converts all filenames to lowercase when sorting them prior to loading a file.  Thus, you essentially end up with duplicate files.  You were selecting `CE2_mount.gcode`, Moonraker was telling Klippy to print it and Fluidd was displaying its thumbnail.  However Klipper was selecting `CE2_Mount.gcode` as the sorting algorithm doesn't handle duplicates.

markwinger, 2021-05-08:
I see.  So it is a bug somewhere.  I selected the file with fluidd using upload/print.  I would expect that file to print no matter what was uploaded previously.

You said that klipper sorting algorithm doesn't support  duplicates.  But 2 files differing only by case are not duplicates.  Seems to me that either all pieces handle case, or conversion to upper or lower is done so duplicates cannot be created.

I think modifying klipper to fix this would be best, but alternately, fluidd (or moonraker) could convert all file names being uploaded to upper or lower so no duplicates are possible.  I would rather overwrite the old file with the new on instead or reprinting the old one.

Arksine, 2021-05-08:
Someone PRed a patch for this to Klipper, but the patch is wrong.  I'll submit a modified patch that should fix this.

Arksine, 2021-05-08:
Ok, submitted PR KevinOConnor/klipper#4275 which should resolve this issue.

markwinger, 2021-05-09:
Ok, I'll watch for the next klipper update and test it.

Get BlueMail for Android<http://www.bluemail.me/r?b=16361>

markwinger, 2021-05-10:
I created a test case and confirme that case sensitivity was the cause.   Now just waiting for the  commit to show up on an update and I will test again.

cadriel, 2021-05-12:
Fantastic! Thanks all for the resolution here, closing.

closed 2021-05-12 by cadriel