fluidd-core/fluidd #279: Full Screen webcam feed

PigeonFX, 2021-05-09:
## Is your feature request related to a problem? Please describe.
(A clear and concise description of what the problem is. Ex. I'm always frustrated when [...])

## Describe the solution you'd like:
Double-clicking on the webcam to go full screen rather than a 640x480 with maybe an emergency stop button.  
## Describe alternatives you've considered:
(A clear and concise description of any alternative solutions or features you've considered. Again, if possible, think about why these alternatives are not working out.)

## Affected users and/or setups:
I monitor my printing on phone and tablet, so would find having a full size screen of the stream very handy and a quick way to trigger emergency stop.
## Additional Context:
(Add any other context or screenshots about the feature request here.)

The current system of being able to bring up the stream smaller than the one already on the screen seems a little redundant.

[image]

StuSerious, 2021-06-07:
hey there @PigeonFX!

this idea has been discussed and already has backend attention!
expect it in future releases!

cheers,
~S

maruhe, 2021-07-03:
this is the point where I currently (have to) use OctoPrint
(with the 'Fullscreen Plugin') a double-click on the camera-image switches to this view where I have
*) camear in full-window-sice
*) information from printing-widget
*) pause-button (may with extra confirm)
*) full-screen-button to switch between full-window and full-screen

[image]

pedrolamas, 2022-02-20:
On the next release we will provide a button to open the video feed in a new (separate) browser window/tab.

That will only have the feed, no buttons to control the print itself, so for that reason we will keep this ticket open to revisit going forward.

closed 2022-06-16 by matmen