fluidd-core/fluidd #282: Define moonraker api url in config.json

Ramalama2, 2021-05-12:
Please check the screenshot, this is driving me nuts.

The Problem is, fluidd is working perfectly, just i have to enter always the api url of moonraker.
Fluidd/Moonraker are on the same host. Im just accessing it from different devices in the local network, thats whats an webinterface is for in theory xD

[image]

However, the only difference is, that my fluid url is:
https://3d.xxxxx.de:3002/

and i have always to enter as api url: https://3d.xxxxx.de:3002/
is it because fluidd runs on port 3002?

Thank you xD

Ramalama2, 2021-05-12:
I think the main issue is, that that url is saved as a cookie.
And if it is possible to add that url as a config option in fluidd somewhere (config.yaml in the www root for example)

Thanks :-)

cadriel, 2021-05-12:
Im not sure I understand the problem.

Here's some things to keep in mind;

1. When you enter an API URL, it is saved to your browsers local storage for that host URL.
2. As per above, you should never be asked again for that browser and host URL.

Given the above.. if you move to a different browser, you'd have to enter it again.

Ramalama2, 2021-05-13:
> Im not sure I understand the problem.
>
> Here's some things to keep in mind;
>
> 1. When you enter an API URL, it is saved to your browsers local storage for that host URL.
> 2. As per above, you should [...]

exactly, i have multiple browsers and multiple computers and phone/tablet etc.
And they all save the coockie, except if you clear the cache etc...

Aside if that, what is the benefit of it, if it saves into the browser, when you have anyway only the one printer and always the same api url...
i mean if you host fluid for multiple users, that makes sense.
but if you have it locally at home, an option to set into the config file a "default url" would be awesome 🙈

cadriel, 2021-05-14:
There's already a default, this should be preset for you if the host is the same as your moonraker url.

If you have more of a unique setup, then you'll need to rely on the local storage setup. I don't believe under normal circumstances you should be clearing your cache that often.

Closing as working as intended.

Thanks!

closed 2021-05-14 by cadriel

Ramalama2, 2021-05-14:
the host is the same, the port is just different. because of 3 gui's.
octoprint/fluidd/moonraker

Preisschild, 2022-04-12:
fyi, in case anyone is interested, you can do this in a hack-ish if you have nginx as a reverse proxy in front of fluidd.
Basically, this just overwrites the localstorage servers every time the page is loaded.

```
  proxy_set_header Accept-Encoding "";
  sub_filter '<head>' '<head> <script>localStorage.setItem("appInstances", \'[{"apiUrl":"https://<url>","socketUrl":"wss://<url>/websocket","active":true,"name":"<name>"}]\')</script>';
```

matmen, 2022-04-12:
This should have been fixed by https://github.com/fluidd-core/fluidd/pull/583. Is the issue still persistent?