fluidd-core/fluidd #283: Forget thumbnail image when moving a file to a subdir

tremor, 2021-05-12:
## Affected Version(s):
(The versions of fluidd this issue occurs on.)
- klipper version: v0.9.1-492-g60352f4b
- moonraker version: v0.6.0-38-gd3c251b
- fluidd version: v1.13.0
- 

## Platform:
Raspberry Pi 3

## Hardware Configuration:
(Information about your hardware configuration. Like MCU, host and setup)

## Reproduction Steps:
1. Create subdir in fluidd
2. Upload ufp file from Cura to Fluiddpi - thumbnail image is visible in the job status (and there are 2 .png files in the ~/gcode_files/thumbs dir)
3. Drag&Drop file to the subdir
4. Enter subdir - no thumbnail is visible, in the filesystem are no thumbs dir in the subdir and the 2 .png files are delete in the ~/gcode_files/thumbs dir)

## Expected Results:
The files in the ~/gcode_files/thumbs dir should move to ~/gcode_files/subdir/thumbs when Dragging in a subdir

Arksine, 2021-05-12:
This should be resolved when Arksine/moonraker#159 is merged.

photocromax, 2021-05-23:
thumbs files are not regenerated when overwriting a file using Jobs page.

- fluidd v1.14.0
- moonraker v0.6.0-122-ga18987a  ( [Arksine/moonraker#159](https://github.com/Arksine/moonraker/pull/159) already merged )

tremor, 2021-05-25:
This bug is now fixed! Thanks!

closed 2021-05-25 by tremor

photocromax, 2021-05-28:
> This bug is now fixed! Thanks!

It seems it's not fixed.
I added a comment to this issue cause I thought it could be related to the same problem

### How to replicate the issue

1. Slice a model with Prusa slicer (thumbnails activated)
2. Export the GCode file
3. Upload it to fluidd jobs tab
4. Modify (or just re-orient) the model in Prusa slicer
5. Export the GCode file with same name as the previous one
6. Upload it to fuidd job tab

The old GCode is correctly overwritten by the new file.
The previous thumbnail is NOT recalculated (or simply not overwritten).

### Extra info
If the file is uploaded with a different name, the thumbnail generated is correct