fluidd-core/fluidd #284: Macros not showing up

zenturacp, 2021-05-13:
## Affected Version(s):
(The versions of fluidd this issue occurs on.)
- klipper version: v0.9.1-497-gb4a22297
- moonraker version: v0.6.0-38-gd3c251b
- fluidd version: v1.13.0

## Platform:
Windows 10, Edge (Chromium)

## Actual Results:
When adding macros there is just always 1 macro no matter what i do (CANCEL_PRINT) even if i remove it it does not get removed from the interface

## Screenshot(s):
[image]

## Expected Results:
When i add new macros they will get into the interface

## Log Files:
[klippy.log](https://github.com/cadriel/fluidd/files/6474841/klippy.log)
[moonraker.log](https://github.com/cadriel/fluidd/files/6474843/moonraker.log)

## Additional information
Its a clean install of FluiddPi, just installed today with a default print.cfg which includes the other files

cadriel, 2021-05-14:
I can't replicate this. Looking at your screenshot, it suggests your printer is unable to connect. Your macros aren't going to show changes until you resolve that issue.

If you continue to have issues, please feel free to join the discord - we'll be able to better provide support there.

Thanks!

closed 2021-05-14 by cadriel

zenturacp, 2021-05-15:
You are right - aparantly the macros are not populated until connected.. 

Maybe there should be just a small '?' Explaining it?

Or just make it so the CANCEL_PRINT is not shown..?