fluidd-core/fluidd #285: Tool Widget - Touch UI Option

KurioHonoo, 2021-05-14:
## Is your feature request related to a problem? Please describe.

I get frustrated when using Fluidd on my phone because when attempting to scroll down the dashboard the Speed or Flow slider grabs the touch instead and cranks either of the values to about 200% or 2%.

## Describe the solution you'd like:

I propose adding in the ability to swap between sliders and touch buttons similar to the image below. This UI is see on Vez3D's touch screen on his printer, but I think something similar could be worked into the tool widget to be an improved option for touch screen users. 
[image]

## Describe alternatives you've considered:

The alternatives I've tried is remembering to have the tool widget minimized when on mobile as to not accidentally touch the slider or just not using Fluidd on mobile. 

## Affected users and/or setups:

I think that it will benefit everyone, especially those using a mobile device or tablet as there main way of controlling their printer.

## Additional Context:

I added an image to the proposed solution section.

PS. - Thanks for being such an awesome dude.

cadriel, 2021-05-14:
Are you on the latest version of Fluidd? 1.13 introduced a lock on sliders in mobile format to alleviate this issue.

KurioHonoo, 2021-05-14:
No, not yet due to the need to reflash MCUs. A slider lock sounds amazing, but I still feel that the Touch UI request is still valid.

matmen, 2022-05-09:
Hey, just wondering, can this issue be closed considering the slider lock feature?

KurioHonoo, 2022-05-09:
I thought that it had already been because of the slider locks, so go for it.

closed 2022-05-09 by matmen