fluidd-core/fluidd #287: Terminal scrollbar is drawn over long lines

wlhlm, 2021-05-15:
## Affected Version(s):

- klipper version: master branch, commit KevinOConnor/klipper@7b170d8c3dee752bb8c6ee8de730fc0bf6290c5e
- moonraker version: master branch, commit Arksine/moonraker@490e66fe075706265f7264481ef6296bc2d32da7
- fluidd version: develop branch, commit a81c1b76a6cb7951f642f7690a01ee932a8ce33e

## Platform:
- Windows 10 20H2
- Firefox 88.0

## Hardware Configuration:
- BTT SKR Mini E3
- RPi 4 2GB

## Reproduction Steps:

1. Run command in the terminal that produces long response lines
2. Try to read output

## Actual Results:

The terminal scrollbar is drawn over long lines and makes reading them difficult.

## Screenshot(s):
[image]

## Expected Results:

Terminal scrollbar is never drawn over the terminal content

## Log Files:

*Log files omitted.*

cadriel, 2021-05-25:
The problem here isn't that they're long lines, it's that they're long lines with no white space.

I've updated the develop branch to allow a horizontal scroll in this case, hopefully that fixes the issue.

cadriel, 2021-05-25:
Please comment here if this is still an issue after the next release (1.15)

closed 2021-05-25 by cadriel

wlhlm, 2021-05-26:
Testing out commit, the terminal now allows reading the rest of the lines by making by allowing horizontal scrolling.

[image]

> The problem here isn't that they're long lines, it's that they're long lines with no white space.

No, this is not the issue. Take a look at the lines in the console I posted above, if I scroll all the way to the right, you can see that the lines have plenty of white space to allow text wrapping:

[image]

From my understanding, this happens, because the terminal content always keeps the same width regardless of whether the Fluidd column in which the console card is positioned has to be shrunk in order to accommodate for a narrower browser window.

When I increase the width of the browser view port, the line wrapping works without the need for horizontal scrolling:

[image]

My expectation would be that Fluidd rewraps the console contents when the console card width changes.

cadriel, 2021-05-26:
> Testing out commit [dea2c1d](https://github.com/cadriel/fluidd/commit/dea2c1dd34a4d7120b0179068b1ed9f87fcdce17), the terminal now allows reading the rest of the lines by making by allowing horizontal scrolling. Thanks for taking a look at my report.
> 
> > The problem here isn't that they're long lines, it's that they're long lines with no white space.
> 
> No, this is not the issue. Take a look at the lines in the console I posted above, if I scroll all the way to the right, you can see that the lines have plenty of white space to allow text wrapping:

It is. :) I can't reproduce the issue you're seeing at all when I have standard output, with spaces. I can forcefully reproduce it when I remove whitespace from the content. I've confirmed this on Firefox, and Chrome - on both windows and mac.

My suspicion is the content you're seeing output in your example doesn't have white space (even tho it looks like it does).

Note - the original outout from Klipper runs through a parser to introduce breaks in html where necessary, so sometimes what you see as an automated line break, is not.

> 
> From my understanding, this happens, because the terminal content always keeps the same width regardless of whether the Fluidd column in which the console card is positioned has to be shrunk in order to accommodate for a narrower browser window.
> 
> When I increase the width of the browser view port so that the card is at its maximum width, you can see the line wrapping work (without the need for horizontal scrolling):
> 

Attached a MOV to show things wrap properly for me. I see this result everywhere there's actual white space.

https://user-images.githubusercontent.com/205520/119679303-defa6d80-be05-11eb-8c38-e21faa7e64de.mov

Out of interest, how are you producing the output in your screenshots? If I can reproduce the same output - I'd have a better idea on what you're seeing.

Also, you didn't say - did the change fix the issue for you, or no?

wlhlm, 2021-05-26:
> Attached a MOV to show things wrap properly for me. I see this result everywhere there's actual white space.

That is indeed the behavior I was looking for.

> Out of interest, how are you producing the output in your screenshots? If I can reproduce the same output - I'd have a better idea on what you're seeing.

I'm using the following macro to dump the `printer` variable:

```ini
# taken from https://github.com/zellneralex/klipper_config/blob/04db4bbce9fbbfc4716f15aa703d505509c827fb/macro.cfg#L643-L658
[gcode_macro DUMP_PARAMETERS]
gcode:
    {% set parameters = namespace(output = '') %}

    {% for name1 in printer %}
        {% for name2 in printer[name1] %}
            {% set donotwant = ['bed_mesh','configfile'] %}
            {% if name1 is not in donotwant %}
                {% set param = "printer['%s'].%s = %s" % (name1, name2, printer[name1][name2]) %}
                {% set parameters.output = parameters.output +  param + "\n" %}
            {% else %}
                {% set param = "printer['%s'] = %s" % (name1, printer[name1]) %}
                {% set parameters.output = parameters.output +  param + "\n" %}
            {% endif %}
        {% endfor %}
    {% endfor %}
    {action_respond_info(parameters.output)}
```

Testing using some other macros, I realize that I didn't fully understand the wrapping behavior before, but it makes sense now: the width of the terminal contents is based on the longest character sequence without whitespace. This has nothing to do with the default column width. Everything I said in my [response](https://github.com/cadriel/fluidd/issues/287#issuecomment-848646912) can be ignored.

Here is a test macro that made it click for me:

```ini
[gcode_macro LONG_LINE]
gcode:
    {action_respond_info("abcdefghijklmnopqrstuvwxyz0123456789 abcdefghijklmnopqrstuvwxyz0123456789 abcdefghijklmnopqrstuvwxyz0123456789")}
    {action_respond_info("abcdefghijklmnopqrstuvwxyz0123456789abcdefghijklmnopqrstuvwxyz0123456789 abcdefghijklmnopqrstuvwxyz0123456789")}
    {action_respond_info("abcdefghijklmnopqrstuvwxyz0123456789abcdefghijklmnopqrstuvwxyz0123456789abcdefghijklmnopqrstuvwxyz0123456789")}
```

> Also, you didn't say - did the change fix the issue for you, or no?

My original issue, that some parts of long lines where inaccessible ("hidden behind the scrollbar") is fixed. Thank you.

cadriel, 2021-05-26:
Ok, I think we can safely call this done then.

Thank you! :)