fluidd-core/fluidd #289: Example Reverse Proxy config using Swag

noobydp, 2021-05-23:
Hey Craig, 

I use Swag as a reverse proxy for a bunch of stuff and I wanted to share a sample config file that works with fluidd [fluidd.subdomain.conf.txt](https://github.com/cadriel/fluidd/files/6527539/fluidd.subdomain.conf.txt)case you wanted to add it to the wiki/docs.

I run swag in docker on unraid: https://docs.linuxserver.io/general/swag

Just need to edit the Fluidd IP Address

sample config file stored in `nginx/proxy-confs/fluidd.subdomain.conf`
(i had renamed the .conf file to .txt to attach to this ticket)

[fluidd.subdomain.conf.txt](https://github.com/cadriel/fluidd/files/6527540/fluidd.subdomain.conf.txt)

hofleverancier, 2021-05-23:
Bookmarked, thanks bro.

cadriel, 2021-05-24:
Hi! Thanks for the submission.

I'll take a look and figure out how we can incorporate submissions like this into the docs.

Cheers

--
Craig.

kylerw, 2021-09-02:
@noobydp are you also reverse proxying your moonraker API to accomplish this or can you share how the public fluidd is accessing moonraker? I've tried to do it hitting the local IP address and it's failing so I'm assuming I need to proxy that as well....

noobydp, 2021-09-05:
@kylerw No I didnt have to expose moonraker. For the printer address I used the external https url (i.e. https://fluidd.mydomain.com) when prompted in the Fluidd UI.

I had that problem the first time I tried to get this working (in 2020). I cant remember if an update to fluidd/moonraker fixed it, or it it was just a bad proxy config. Note I could never get this to work as a sub-folder, only a sub-domain.

kylerw, 2021-09-06:
@noobydp  is that running Klipper and Moonraker on the pi and Fluidd on docker? Or is Fluidd also on the pi?

noobydp, 2021-09-07:
@kylerw Everything on the pi

ste94pz, 2021-12-17:
Hi, do you know how to configure the reverse proxy with Apache? I'm trying but doesn't work, maybe because the links inside the fluidd files are absolute and not relative.

noobydp, 2021-12-17:
@ste94pz sorry I don’t, Apache always seemed like complex overkill for this. 
I run Nguni using the LinuxServer Swag docker container https://docs.linuxserver.io/general/swag
(And Authelia for authentication)

closed 2021-12-17 by noobydp

rgnd971, 2023-10-14:
Great @noobydp , it works for me on my OMV nas via swag! However, the video stream does not appear.