fluidd-core/fluidd #290: camera fps/res

mironfs, 2021-05-24:
i tried to edit fluiddpi.txt to try some other resolutions and fps but its still 640x480 10 fps. 
i tried -fps and -f too, i tried -r and -x -y and still nothing. i tried a lot of command like this https://3dp.tumbleweedlabs.com/firmware/klipper-firmware/adding-webcam-support-to-mainsail#installing-mjpg-streamer but i think it needs to stop original process first

[fluiddpi.txt](https://github.com/cadriel/fluidd/files/6534712/fluiddpi.txt)

mironfs, 2021-05-24:
i use ir raspicam from aliexpress

cadriel, 2021-05-24:
Hey! Thanks for the report. This sounds more like a support request, so I ask you come join us on Discord in order to get help.

Changes in the `fluiddpi.txt` require a host reboot to take effect, and some config changes don't apply depending on the camera type you're using.

Note that the FPS display in Fluidd represents your ACTUAL FPS, versus what you're requesting - because often your wifi network bandwidth from the Pi itself restricts your available FPS.

Thanks, here's the link to Discord;
https://discord.gg/GZ3D5tqfcF

closed 2021-05-24 by cadriel