fluidd-core/fluidd #294: Key '[*]' in namespace 'history' not found

DEAth1981, 2021-05-29:
I turned on the printer, opened the Fluidd web interface and received an error (see screenshot).
If you delete the [history] section from moonraker.conf, the error no longer appears, but then there is no History section on the Dashboard.
How to fix this?

## Affected Version(s):
- klipper version: 0.9.1-525-g8e74e6aa
- moonraker version: 0.6.0-136-g793ab6b
- fluidd version: 1.14.0

## Platform:
WIndows 10 Corporation LTSC x64 (v.10.0.17763.1935)
Opera v.10.0.17763.1935

## Hardware Configuration:
MCU - BTT GTR v1.0
Host - Raspberry Pi 3B+

## Screenshot(s):
[image]

## Expected Results:
Errors should not be

## Log Files:
- [klipper log](https://github.com/cadriel/fluidd/files/6565204/klippy.log)
- [moonraker log](https://github.com/cadriel/fluidd/files/6565206/moonraker.log)

Arksine, 2021-06-01:
The `moonraker.log` you submitted has `[history]` excluded.  To debug this I'll need you to add `[history]` to `moonraker.conf`.  It would also be helpful if you could add the following to the `[server]` secition:
```
enable_debug_logging: True
```

Then reproduce the error and re-submit the log.  Afterwards I suggest removing the debug logging option, as it can cause logs to grow quite large.

cadriel, 2021-06-07:
We've figured this out. The latest moonraker release should resolve these issues.

Please feel free to comment here if you still encounter the error after updating.

Thank you!

DEAth1981, 2021-06-07:
I can’t provide information yet, because the STM-chip on my board burned out, the printer was disassembled, I’m waiting for the mcu to be repaired.

DEAth1981, 2021-06-11:
> The `moonraker.log` you submitted has `[history]` excluded. To debug this I'll need you to add `[history]` to `moonraker.conf`. It would also be helpful if you could add the following to the `[server]` section:
> 
> ```
> enable_debug_logging: True
> ```
> 
> Then reproduce the error and re-submit the log. Afterwards I suggest removing the debug logging option, as it can cause logs to grow quite large.

[moonraker.log](https://github.com/cadriel/fluidd/files/6640889/moonraker.log)

cadriel, 2021-06-11:
Did you update moonraker, and still get the error?

DEAth1981, 2021-06-11:
> Did you update moonraker, and still get the error?

Yes, the error remains

cadriel, 2021-06-11:
@Arksine Seems like our fix may not have resolved this one.

Arksine, 2021-06-12:
A log reproducing the issue is going to be necessary to troubleshoot it.

DEAth1981, 2021-06-12:
> A log reproducing the issue is going to be necessary to troubleshoot it.

2021-06-11 23:36:35,171 [websockets.py:build_error()] - JSON-RPC Request Error: 404
Key '['']' in namespace 'history' not found
Traceback (most recent call last):
  File "/home/pi/moonraker/moonraker/components/database.py", line 208, in get_item
    key_list = self._process_key(key)
  File "/home/pi/moonraker/moonraker/components/database.py", line 284, in _process_key
    raise self.server.error(f"Invalid Key Format: '{key}'")
utils.ServerError: Invalid Key Format: '['']'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/moonraker/moonraker/websockets.py", line 219, in execute_method
    result = await method(ws, *args, **kwargs)
  File "/home/pi/moonraker/moonraker/websockets.py", line 312, in func
    ip_addr=ws.ip_addr, user=ws.current_user))
  File "/home/pi/moonraker/moonraker/components/history.py", line 165, in _handle_jobs_list
    job = self.history_ns[job_id]
  File "/home/pi/moonraker/moonraker/components/database.py", line 445, in __getitem__
    return self.get(key, default=SENTINEL)
  File "/home/pi/moonraker/moonraker/components/database.py", line 434, in get
    return self.db.get_item(self.namespace, key, default)
  File "/home/pi/moonraker/moonraker/components/database.py", line 215, in get_item
    f"Key '{key}' in namespace '{namespace}' not found", 404)
utils.ServerError: Key '['']' in namespace 'history' not found
2021-06-11 23:36:35,171 [websockets.py:dispatch()] - Websocket Response::{"jsonrpc": "2.0", "error": {"code": 404, "message": "Key '['']' in namespace 'history' not found"}, "id": 72872}

These lines are in moonraker.log, which I posted earlier (several posts above).
What other log is needed?

Arksine, 2021-06-12:
That traceback is not in the log posted above.  The full log with that traceback and history enabled is what is needed.

Arksine, 2021-06-12:
Oh, NM, it might be in the second log, let me check

Arksine, 2021-06-13:
After looking at this its possible that somehow the database got spoiled with an invalid job id. Could you try entering the following url in the browser and post the response (it could be large, if necessary save to the a file and attach):
```
http://192.168.1.20:7165/server/database/item?namespace=history
```
Having an idea of exactly what is in the database I may be able to provide guidance on how to repair or, or possibly automate the repair if I can track down what caused it.

DEAth1981, 2021-06-13:
> After looking at this its possible that somehow the database got spoiled with an invalid job id. Could you try entering the following url in the browser and post the response (it could be large, if necessary save to a file and attach):
> 
> ```
> http://192.168.1.14:7125/server/database/item?namespace=history
> ```
> 
> Having an idea of exactly what is in the database I may be able to provide guidance on how to repair or, or possibly automate the repair if I can track down what caused it.

[database_history.txt](https://github.com/cadriel/fluidd/files/6643659/database_history.txt)

Arksine, 2021-06-13:
Ok, I pushed a series of commits that should repair the database and get things working.  Somehow the database did indeed get spoiled with an invalid key.  After updating Moonraker it will remove the the invalid entry which should allow the list request to return without an error.

DEAth1981, 2021-06-14:
> Ok, I pushed a series of commits that should repair the database and get things working. Somehow the database did indeed get spoiled with an invalid key. After updating Moonraker it will remove the the invalid entry which should allow the list request to return without an error.

Updated Moonraker and instead of fixing [history] got an inoperable Fluidd (got Moonraker killed). :)
In general, I did a complete reinstallation (I reinstalled FluiddPi).

cadriel, 2021-06-15:
Ok, I'm going to close this issue now. Please feel free to comment here if you encounter a similar issue again.

Thank you all!

closed 2021-06-15 by cadriel