fluidd-core/fluidd #295: Feature Request: End Time Display

Mitsuma, 2021-05-29:
## Describe the solution you'd like:
One thing I loved, coming from OctoPrint, was having an addon that showed me the time (as in clock time) when my print will finish.
Right now we only see the remaining time and as silly as it sounds making the effort to calculate the time when it finishes is a little annoying.

Little "Mockup":
[image]
I would like 24h but maybe add an option for AM/PM as well. (Hour/Minutes is probably enough.)
Doesn't need to show the full date but might not be a bad option to add, for multi day prints.

## Describe alternatives you've considered:
No alternatives outside of going back to OctoPrint.

## Affected users and/or setups:
Probably a good help for everybody and not just me who is too lazy to calculate the end time.

## Additional Context:
(Add any other context or screenshots about the feature request here.)

cadriel, 2021-06-06:
Hi, thank you! I'll look into this with the next status card update.

kieraneglin, 2021-06-14:
@cadriel I just discovered this project and the Vue stack is right in line with what I'm most familiar with.  I'm interested in contributing and would love to take this on as my first issue if it hasn't already been implemented!

cadriel, 2021-06-14:
Yep, we love contributions. Please feel free to join us on Discord to discuss.

https://discord.gg/GZ3D5tqfcF

cadriel, 2021-06-19:
This is now in place for next release. Thanks all!

closed 2021-06-19 by cadriel