fluidd-core/fluidd #304: v1.14.0-e46475e : app.fluidd.xyz hangs during typing the URL

Drachenkaetzchen, 2021-06-06:
## Affected Version(s):
(The versions of fluidd this issue occurs on.)
- klipper version: Not relevant
- moonraker version: Not relevant
- fluidd version:  v1.14.0-e46475e (via app.fluidd.xyz)

## Platform:
- OS not relevant
- Happens in Chrome as well as in Firefox

## Hardware Configuration:

Not relevant

## Reproduction Steps:
1. I started typing the URL of my moonraker installation at `https://app.fluidd.xyz/`. The full URL is `http://voron-klipper.feli-home.drachenkatze.org` (not reachable from the internet) and the issue occurs as soon as I have entered `http://voron-klipper.feli-home.drachenkatze`.

## Actual Results:
The browser hangs. On Chrome, the tab becomes unresponsive indefinitely. On Firefox, the page hangs for quite a while and I can see several JavaScript errors in the console.

## Screenshot(s):
[image]

## Expected Results:
No hang.

## Log Files:
Both not relevant, as no connection is attempted.

Drachenkaetzchen, 2021-06-06:
I did test using copy'n'paste of the URL, this doesn't hang the browser.

I don't think fluidd should try a HTTP request as soon as one key is pressed ;)

cadriel, 2021-06-06:
I can't reproduce the recursion issue you're seeing. Fluidd doesn't start pinging anything until it see's what it thinks is a valid URL, at which point (with a debounce) it will ping the given URL in order to determine if it's a valid moonraker instance.

In your screenshot above - your actual issue is one of mixed content, in that you're likely hitting `https://` in your browser - and then trying to connect to a moonraker instance on `http://`. This is a valid security concern in browsers, and so is denied. 

If you fix the protocol issue, I suspect it'll start working for you.

Independently of that, If you have other detail on how to reproduce the recursion issue - I could look into that.

Drachenkaetzchen, 2021-06-06:
I tried with http://, and it's the same problem when I started typing the URL. However, I could get a connection when I copy'n'pasted the full URL.

I assume that it tries multiple times, because when I'm at home, `http://voron-klipper` reaches the instance since my DNS suffix is `feli-home.drachenkatze.org`, so would `http://voron-klipper.feli-home` and `http://voron-klipper.feli-home.drachenkatze`.

cadriel, 2021-06-07:
Ok, so I suspect this is a result of the fetch not being cancelled as new strings are typed. I've added a fix for this for the next release. Let me know how you get on once that happens.

Thanks!

cadriel, 2021-07-01:
Is this resolved with the latest release?

Drachenkaetzchen, 2021-07-07:
Unfortunately no. This time, the whole tab hangs with no further activity.

[image]

cadriel, 2021-07-07:
The error you're showing above still has the mixed content issues - and so the error in the dialog is correct.

 I also haven't heard from anyone else that's having a similar issue.

Adding a test button when we can do it automatically isn't ideal.

I notice that there are some line items in your console that have nothing to do with Fluidd, which can be a cause for concern. Perhaps disable your extensions to rule out anything at play there?

Otherwise, the errors in that console pic are all relevant and accounted for.

cadriel, 2021-07-07:
Does it work if you load http://app.fluidd.xyz instead of https://app.fluidd.xyz? :)

Drachenkaetzchen, 2021-07-07:
> Does it work if you load http://app.fluidd.xyz instead of https://app.fluidd.xyz? :)

No. No matter if https or http, it hangs completely in Chrome after I typed `http://voron.feli-home.drachenkatze`. In Firefox, the same recursion error as before occurs.

[image]

> I notice that there are some line items in your console that have nothing to do with Fluidd, which can be a cause for concern. > Perhaps disable your extensions to rule out anything at play there?

The messages in the console have nothing to do with the hang - they appear before the hang.

Here's a video recording of what's happening. Immediately after typing `http://voron.feli-home.drachenkatze` the window becomes unresponsive around 0:14). At 0:26 the recursion error would've been reported in the console, and the letters `org` finally appear although already typed at 0:14.

https://youtu.be/02H9uu9rq2Q

Drachenkaetzchen, 2021-07-07:
Another observation: I just tested it with disconnecting from WLAN after loading `app.fluid.xyz`, the same issue occurs. No HTTP request is being made, so it's not even a network issue.

cadriel, 2021-07-07:
Ok, I've partially replicated the issue. I can get the window to hang for a bit, but only when manually typing the URL you've given above - and I don't see the recursion errors in the console.

However - I can't replicate it on the development builds - of which I've changed some of the logic surrounding this. I suspect its some kind of weird recursion issue in the regex which I've since dropped.

I'll also do a double take on the throttle being used here too - and hopefully, with the next release - we'll get this sorted.

Thanks for bearing with me here - this is a really weird bug.

Drachenkaetzchen, 2021-07-07:
No worries. I can work around it by copy'n'paste the URL, however, other users might not realize that this works around the issue.

cadriel, 2021-07-08:
This is now release in 1.16.0! Can you confirm if this resolves your issue? Thanks!

Drachenkaetzchen, 2021-07-11:
I can confirm that the issue is now fixed. Thank you for the effort!

closed 2021-07-11 by Drachenkaetzchen