fluidd-core/fluidd #307: Chrome/ium is now default enabling CORS-RF1918

sreichholf, 2021-06-10:
## Affected Version(s):
(The versions of fluidd this issue occurs on.)
- klipper version: v0.9.1-550-gecbfa762
- moonraker version: v0.6.0-148-gd9d3739
- fluidd version: app.fluidd.xyz

## Platform:
Chrome 91.0.4472.101

## Reproduction Steps:
1. Install Chrome or Chromium >= 91
2. Try to use app.fluidd.xyz

## Actual Results:
The Browser will happily report something in the form of
``Access to XMLHttpRequest at 'http://klipper.local/server/database/item?namespace=fluidd&key=layout' from origin 'http://app.fluidd.xyz' has been blocked by CORS policy: The request client is not a secure context and the resource is in more-private address space `private`.
``

## Expected Results:
No error ;)
my.mainsail.xyz still works, but i haven't had a look into why it does.

## Additional information
in the cors_domains: part of mooraker
    http://app.fluidd.xyz
is present.

There actually is a flag on chrome allowing to disable that change and make fuild.xyz work again, but it wouldn't recommend doing so.
The flag is: chrome://flags/#block-insecure-private-network-requests

cadriel, 2021-06-11:
RFC1918 doesn't come into effect until Chrome 92, scheduled for sometime in July.
https://chromestatus.com/feature/5436853517811712

Given my environments, all seems to working as expected my side. I'd assume its because you're using Chromium - and they may have enabled the flag early.

Mainsail also uses the same connection method, and has the same restrictions Fluidd will have when the RFC comes into effect. I'm not sure why you're seeing different results.

The only way this is going to work moving forward is by serving your moonraker instance on https. It's unfortunate, because this isn't something many users are going to find easy to do.

Moonraker also now supports HTTPS - however, generation of the certificate is going to be up to the user until we can implement some automation around this.

DanielJoyce, 2021-06-14:
Adding the self-signed cert to your browser isn't hard, and can be documented.

cadriel, 2021-06-14:
Yes, we will likely do this. But its less ideal from a users perspective, I'd prefer it "just works".

Ramalama2, 2021-06-19:
> Adding the self-signed cert to your browser isn't hard, and can be documented.

Hi, self signed certs doesn't work anymore either in newer firefox releases 🤦

But im running here fluid with https and that's super simple with nginx etc...

I see the only problem for people is to get actually a domain, to be able to get a certificate... Other as that, everything can be automated with a script or whatever.

But those who can't get a domain, for whatever reason, can use app.fluidd.xyz ? 🙈

Whatsoever, I don't even understand why we need certificates for rfc1918 (private networks). If someone is able to access your network, and fake or spoof a new mac address to redirect you to the fake website, in your private network environment....
The last problem you have is the ssl certificate, lol
And for mdm attacks in your own network, I don't want even to start.
However, that rfc should be optionally, but not mandatory.
Just stupid 😂 but sorry, that's off topic 🙈

cadriel, 2021-06-19:
Its worth noting we have a ticket with the Chrome team about this, and they've pushed back the implementation until WebTransport is implemented in Chrome.

WebTransport is the recommended approach to resolving the particular issues we face.

You can see the outcome of that here;
https://docs.google.com/document/d/1bpis0QwaA9ZrRFmpPW6LiaPmdwT0UhhUMNsEnU0zfLk/edit#heading=h.cjkkjbd3h2lo

For now, I'll close this ticket - with the knowledge that we're aware of the upcoming changes - and are working through possible solutions.

Thanks everyone!

closed 2021-06-19 by cadriel