fluidd-core/fluidd #310: Add support for UV4L-MJPEG (IP Camera) Support

rizwansarwar, 2021-06-12:
## Is your feature request related to a problem? Please describe.
No
## Describe the solution you'd like:
Mainsail recently added support for UV4L-MJPEG i-e IP camera support. This will be great addition to allow webcam for IP Cameras such as Yi/Waze etc. 

## Describe alternatives you've considered:
None
## Affected users and/or setups:
None
## Additional Context:
None

cadriel, 2021-06-13:
Doesn't this just stream WebRTC, and therefor require the html5 <video> tag? In which case - Fluidd has supported this for a while now.

rizwansarwar, 2021-06-13:
The stream is certainly WebRTC compliant, there is some pre-reqs such as installing the uv4l-mjpegstream plugin and then setting it up so that it can convert any IP Camera sending http stream to WebRTC. The config creates a dummy device that can then be read as if the camera is connected to Pi. The support requested should check for the presense of the plugin and then allow the uv4l config to done via web interface when adding camera.

cadriel, 2021-06-13:
I'll need to spend some time testing this to determine what the actual front-end requirements are to set it up.

ChaosBlades, 2022-08-29:
Wouldn't it be simpler to integrate this?
https://github.com/fluidd-core/FluiddPI/issues/51

pedrolamas, 2022-08-30:
> Wouldn't it be simpler to integrate this? [fluidd-core/FluiddPI#51](https://github.com/fluidd-core/FluiddPI/issues/51)

At this stage, I think the path forward might be just to integrate with [crowsnest](https://github.com/mainsail-crew/crowsnest).

ChaosBlades, 2022-08-30:
I didn't know about that. 

Personally I don't care what the 3D printing community does concerning cameras as long as it involves moving away from the absolutely terrible mjpeg streams and towards h264.

pedrolamas, 2022-12-16:
For all matters and purposes, Fluidd web interface already supports most types of camera, as long as the browser has support for it too... we even support setting an http page for which we use an `<iframe>`!

I don't see what more we can do on this one, so will be considering this complete and closing the ticket.

closed 2022-12-16 by pedrolamas