fluidd-core/fluidd #313: FR: Add button to the screen for built in klipper macros

DanielJoyce, 2021-06-14:
I've searched and not seen the option, but it would be nice if we could add buttons to the screen that emit single g code commands, such as for manual bed leveling. Right now, we need to use the terminal, or define a 'custom macro' which all it does is wrap a built in macro/gcode.

DanielJoyce, 2021-06-14:
Specifically, I'd like to have this for manual bed adjust, which doesn't seem to be a feature in fluidd yet?

StuSerious, 2021-06-14:
hey there @DanielJoyce!

I recently discussed something similar with Cadriel, and it's going to be in the pipeline.
no ETA yet, but it will eventually make its way into fluidd!

as a quick reminder, PRs are always welcome <3

cheers,
~S

fuzzynm, 2021-06-18:
If your looking just for manual bed leveling, define  [bed_screws], and the button BED_SCREW_ADJUST shows up on the tool window.

lar3ry, 2021-07-12:
Not quite sure what you mean. Right now, if you put a macro into your printer.cfg, it will show up in the Macros box.
For example, I have three macros for manual be levelling, called Lev_LF, Lev_RF, and Lev_BK (I like to keep the names as short as possible). I also made another macro "folder" so I don't have to use or see the macros supplied with Fluidd.

You should have no reluctance about making macros in printer.cfg, even if they consist of just one GCode. You can make them visible as buttons or not.
Here's what I have for Lev_RF:

[gcode_macro Lev_RF]
gcode:
 G28
 G1 Z5
 G1 X290 F10000
 G28 Z

StuSerious, 2022-01-30:
We since went back and worked a bit with the macros, I would consider this issue resolved.

@DanielJoyce do you agree?

lar3ry, 2022-01-30:
On Sun, 2022-01-30 at 11:59 -0800, Stefano Serioli wrote:
> We since went back and worked a bit with the macros, I would consider
> this issue resolved.
> 
> @DanielJoyce do you agree?

Yes.

closed 2022-02-18 by pedrolamas