fluidd-core/fluidd #317: [v1.14.0] Camera feed is not stopped in Firefox when not visible

LucidityCrash, 2021-06-15:
## Affected Version(s):
- fluidd version: v1.14.0

## Platform:
(Information about your platform, like the OS or browser you are using.)

## Hardware Configuration:
(Information about your hardware configuration. Like MCU, host and setup)

## Reproduction Steps:
1. (Something you did.)
2. (Something you did next.)

## Actual Results:
(What happens after the reproduction steps have been followed.)

## Screenshot(s):
(Images or gifs showing the problem, perhaps before/after.) 

## Expected Results:
(What *should* happen after the reproduction steps have been followed.)

## Log Files:
(The log files fluidd provides at Configure > System Control.)
- klipper log:
- moonraker log:

## Additional information
(Extra information relevant to the issue, links to similar issues.)

Ramalama2, 2021-06-19:
Hi, im unsure if this has even something todo woth fluidd. The stream gets piped from mjpg-streamer, similar to an iframe.

And additionally to that the browser (changing tabs) behaviour, depends how the browser is setted to keep alive the session.
If you turn off your display, the browser still runs...
If you close the lid, the computer goes to standby or sth similar and the browser isn't active...

However, this doesn't looks for me like anything fluidd related.
Cheers

LucidityCrash, 2021-06-21:
> However, this doesn't looks for me like anything fluidd related.
 
This : 
https://github.com/cadriel/fluidd/issues/32 and https://github.com/cadriel/fluidd/issues/107
would suggest otherwise

Ramalama2, 2021-06-24:
> > However, this doesn't looks for me like anything fluidd related.
> 
> This :
> #32 and #107
> would suggest otherwise

Yes, and i don't know tbh how he does this 🤦
However, good to know, crossing fingers for you 👍

cadriel, 2021-07-03:
I'll take another look at this.

Its a bit hacky the way this is implemented, because I essentially remove the src from the image when either the tab loses focus, or is otherwise collapsed - however, different browsers handle it differently.

The reality is that mjpeg is a less than ideal solution to streaming video tho - so we'll see where I end up.

matmen, 2022-02-25:
I can't reproduce this on either Firefox or Chrome (both on Windows) using adaptive MJPG streaming. Is this still an issue?

matmen, 2022-03-03:
Closing because of staleness. If this is still a problem, feel free to re-open.

closed 2022-03-03 by matmen